KlipperScreen/KlipperScreen #800: [BUG]

dhstsw, 2022-11-21:
**Describe the bug**
Settings are not saved:
If i go in settings and disable DPMS (my screen isn't compatible) and Screen Power Off Time (i usually put a 15 minutes timeout), after next reboot of the host i have to set them again (ence, settings are not saved).

**To Reproduce**
Modify above settings, reboot host.

**Expected behavior**
settings being saved.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logfile**

[KlipperScreen.zip](https://github.com/jordanruthe/KlipperScreen/files/10055931/KlipperScreen.zip)

guestisp, 2022-11-21:
as workaround, ensure that you have a `KlipperScreen.conf` file, even empty, in the config directory.
Without that file, settings are not saved, the file must exist.

alfrix, 2022-11-21:
the log shows
```
2022-11-20 18:46:40,611 [config.py:save_user_config_options()] - Error writing configuration file in /home/pi/klipper_config/KlipperScreen.conf/KlipperScreen.conf:
[Errno 2] No such file or directory: '/home/pi/klipper_config/KlipperScreen.conf/KlipperScreen.conf'
```

i have been testing and i'm unable to reproduce the issue that duplicates the filename
in the meantime yeah you can create blank file named KlipperScreen.conf in printer_data or klipper_config along with your printer.cfg

guestisp, 2022-11-22:
> i have been testing and i'm unable to reproduce the issue that duplicates the filename
> in the meantime you can create blank file named KlipperScreen.conf in printer_data/config or klipper_config

I had the same on my Pi. 
I've tried to fix on my own (that's why i've forked the repo) but honestly I don't know if you would like to use the first path available or if you have any preference in the order.

Anyway, just using this code replicate the issue:
```
import os

filepath = os.path.expanduser("~/")
klipper_config = os.path.join(filepath, "klipper_config")
if os.path.exists(klipper_config):
    filepath = os.path.join(klipper_config, "KlipperScreen.conf")
    
klipper_config = os.path.join(filepath, "printer_data", "config")
if os.path.exists(klipper_config):
    filepath = os.path.join(klipper_config, "KlipperScreen.conf")
else:
    filepath = os.path.join(filepath, "KlipperScreen.conf")

print (filepath)
```

it's due to the latest else, that will join the `KlipperScreen.conf` to the first `filepath = os.path.join(klipper_config, "KlipperScreen.conf")` probably you are missing an if or an else

commit e90dac95 (2022-11-22): config: make klipper_config a secondary location for new configs fixes #800
closed 2022-11-22 by alfrix