KlipperScreen/KlipperScreen #806: Printer selection not working anymore after some iterations

Misterke, 2022-11-22:
**Describe the bug**
After the most recent fixes I can again see a printer selection screen, but after some switches between printers I get stuck that I can no longer select either of my 2 printers on the printer-selection screen.

**To Reproduce**
Unsure, but I have 2 printers configured, both turned off - so with Klipper giving an error - and then just start KlipperScreen with no default printer set.  I then select one printer, when its dashboard appears I click the switch printer icon again, then select the other printer and repeat ...  After a couple of iterations I then get to the printer selection screen, can press either printer (read: see the button getting high-lighted), but that no longer triggers any transition.

The log at that point only gets a single additional 'Closing websocket' print (see below):
```
2022-11-22 20:39:01,939 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
2022-11-22 20:39:01,957 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-11-22 20:39:01,974 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
2022-11-22 20:39:01,975 [KlippyWebsocket.py:connect()] - Starting websocket thread
2022-11-22 20:39:01,998 [KlippyWebsocket.py:close()] - Closing websocket
2022-11-22 20:39:02,009 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2022-11-22 20:39:02,009 [_logging.py:warning()] - websocket connected
2022-11-22 20:39:05,450 [KlippyWebsocket.py:close()] - Closing websocket
2022-11-22 20:39:06,936 [KlippyWebsocket.py:close()] - Closing websocket
2022-11-22 20:39:07,816 [KlippyWebsocket.py:close()] - Closing websocket
2022-11-22 20:39:08,967 [KlippyWebsocket.py:close()] - Closing websocket
2022-11-22 20:39:18,837 [KlippyWebsocket.py:close()] - Closing websocket
2022-11-22 20:39:19,776 [KlippyWebsocket.py:close()] - Closing websocket
2022-11-22 20:39:20,784 [KlippyWebsocket.py:close()] - Closing websocket
2022-11-22 20:39:22,554 [KlippyWebsocket.py:close()] - Closing websocket
2022-11-22 20:40:44,143 [KlippyWebsocket.py:close()] - Closing websocket
2022-11-22 20:40:45,759 [KlippyWebsocket.py:close()] - Closing websocket
```

**Expected behavior**
Always correct working printer selection screen.

Misterke, 2022-11-22:
I think the problem is that in screen.py connect_printer contains:
```
        if self._ws is not None and self._ws.connected:
            self._ws.close()
            return
```
and with adding some extra logging prints I could see that in the error-case _ws.connected is still True and hence KlippyWebsocket's close() is called, which should call close on the actual websocket, but for some reason this does not seem to invoke any on_close callback.

I don't fully understand what should be closing that websocket as I don't get the impression that switching to the select_printer panel even attempts to close an active websocket.  I'm sorry, but I'm not getting to the bottom of this one and will for now switch back to my own fix for the original sigsegv threading issue in [https://github.com/Misterke/KlipperScreen/tree/fix-sigsegv].  If you do find a fix and can finally get the multi-printer selection working again, I'll gladly test it with my setup.

alfrix, 2022-11-22:
the `return` was introduced in 7ca37c56b65c38db53c682c37bf8104e989e2039 to retry closure, but it seems that was a bad idea, i will old behavior while i investigate possible causes of the failure to close

commit 9211583a (2022-11-23): fix issues when using multiple printers fixes #806
closed 2022-11-23 by alfrix