KlipperScreen/KlipperScreen #809: [BUG] No live view of Z-Value with activated Bed-Mesh

Zuendapp, 2022-11-24:
Hello,

in my Ender 3 v2 with Fluidd and KlipperScreen there is no "live" Z-Value shown like in the Fluidd Webinterface.
There ist only the fixed layer hight of Z but no the live Z-Value with adjustments i.e. by a saved bed-mesh like in the Fluid Webinterface visible.
Is there any option to activate this?

Thank you very much!

BR

alfrix, 2022-11-26:
This is not a bug...
the printing panel z is from gcode, it's by design and not configurable at the moment
i was thinking on changing the secondary page that has XYZ to real tooolhead coordinates, so i will close this issue when that gets merged

closed 2022-11-26 by alfrix

Zuendapp, 2022-11-27:
> This is not a bug...
> the printing panel z is from gcode, it's by design and not configurable at the moment
> i was thinking on changing the secondary page that has XYZ to real tooolhead coordinates, so i will close this issue when that gets merged

Perfect, thank you!

commit e9df355b (2022-11-26): job_status: initialize some labels and use live position close #809