KlipperScreen/KlipperScreen #810: [BUG] Klipperscreen not working on 64Bit

Steven2k6, 2022-11-25:
If I change the raspberry to 64Bit, the screen keeps black.

alfrix, 2022-11-26:
Hi, i Can't do anything without logfiles to see the issues

hwmland, 2022-11-27:
This is what helped me (RPi4, 64b):
```shell
sudo apt purge xserver-xorg-video-fbturbo
# consider blacklisting this package by adding:
# Package: xserver-xorg-video-fbturbo Pin: origin "" Pin-Priority: -1
# to the file /etc/apt/preferences
```

Steven2k6, 2022-11-27:
This is the KlipperScreen log, I found nothing wrong.
@hwmland how do I blacklist the package?

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10098670/KlipperScreen.log)

Steven2k6, 2022-11-27:
I changed the GL Driver to Fake KMS and now its showing the SCreen, but it´s streched.

alfrix, 2022-11-27:
The log seems ok, ks is starting and working, stretched screen is the result of not configuring it.
usually it involves adding lines to /boot/config.txt
for example for 800x480:
```
hdmi_group=2
hdmi_mode=87
hdmi_drive=2
hdmi_cvt=800 480 60 6 0 0 0
```
consult the instructions of the manufacturer of the screen for specific info

closed 2022-11-27 by alfrix

Steven2k6, 2022-11-27:
I use an Display called 3.5" RPI Display and I downloaded the software for it and set the resolution of it (480x320). It's connected over spi so I set the SPI-1.

alfrix, 2022-11-27:
you need the hdmi_cvt lines anyway

your log shows it was ok and then you changed it
[image]

Steven2k6, 2022-11-27:
> you need the hdmi_cvt lines anyway
> 
> your log shows it was ok and then you changed it [image]

It´s still streched after setting the resolution in the config.
[config.txt](https://github.com/jordanruthe/KlipperScreen/files/10099234/config.txt)

whiterabbit62, 2023-01-06:
> 

Hi Steven,
I do have the same LCD. It is connected to the Pi zero. It is xpt2046 LCD, 3.5 inch.
I am trying to make it work with Klipper screen on a 32bit mainsailOS, If you know how to make it work, could you let me know the way i can make it work ?

Steven2k6, 2023-01-06:
@whiterabbit62 I tried it for a long time and didn't make it work, so I gave up and bought the btt pitft and it works great and I haven't had any problems with it at all so far.

hwmland, 2023-01-06:
@whiterabbit62 xpt2046 is touch screen controler. Do you have troubles just with touch, or do you have problem with LCD (displaying)? What LCD controler it has if 2nd is true?

Steven2k6, 2023-01-06:
@hwmland Touch isn't working

whiterabbit62, 2023-01-06:
> @whiterabbit62 xpt2046 is touch screen controler. Do you have troubles just with touch, or do you have problem with LCD (displaying)? What LCD controler it has if 2nd is true?

Hi,
It says panel drive IC HX8357D
XPT2046 Touch controller,
This is the product link;
https://cdn.awsli.com.br/945/945993/arquivos/MPI3501-3.5inch-RPi-Display-User-Manual-V1.0.pdf
[image]

LCD works but even if KS is installed it still display the system login info as shown below;
[image]

I assume i have to alter the config file but i have no idea what to do so i did not change anything yet.
Installed KS via kiuh script, all system are updated. This is my first installation KS so i am still newbie.

whiterabbit62, 2023-01-06:
> 

waveshare library does not. You could follow the instruction in the pdf file i shared  then LCD works displaying. But i couldnt make it display Kilipperscreen.

whiterabbit62, 2023-01-06:
> @hwmland Touch isn't working, but I also had issues displaying something.

waveshare library does not. You could follow the instruction in the pdf file i shared then LCD works displaying. But i couldnt make it display Kilipperscreen.

Steven2k6, 2023-01-06:
@whiterabbit62 You can use this [Trobleshooting steps](https://klipperscreen.readthedocs.io/en/latest/).

whiterabbit62, 2023-01-06:
> @whiterabbit62 You can use this [Trobleshooting steps](https://klipperscreen.readthedocs.io/en/latest/).

i have done a manual installation of KlipperScreen then it worked like a charm.

whiterabbit62, 2023-01-06:
> @whiterabbit62 xpt2046 is touch screen controler. Do you have troubles just with touch, or do you have problem with LCD (displaying)? What LCD controler it has if 2nd is true?

Do you have any hint to make touch panel orientation right ?
I have been trying all possible transformation but none of them seems working.
i have tried these
    0°: 1 0 0 0 1 0 0 0 1
    90° Clockwise: 0 -1 1 1 0 0 0 0 1
    90° Counter-Clockwise: 0 1 0 -1 0 1 0 0 1
    180° (Inverts X and Y): -1 0 1 0 -1 1 0 0 1
    invert Y: -1 0 1 1 1 0 0 0 1
    invert X: -1 0 1 0 1 0 0 0 1

whiterabbit62, 2023-01-06:
Okay, if anyone who is having trouble with this type of screen, i am sharing how did i solve the issue about touch panel orientation,
http://www.lcdwiki.com/res/Show_Direction_and_Touch/How_to_change_display_direction-GPIO-Resistive_Touch-V1.2.pdf
As it is shared in this pdf,
I rotated screen 90 degre without swapping axis which means,

Over SSH;
sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./LCD35-show  
---------------------------
cd LCD-show/
sudo ./rotate.sh 0
---------------------------

sudo nano /etc/X11/xorg.conf.d/99-calibration.conf

and editing the content as ;

Section "InputClass"
Identifier "calibration"
MatchProduct "ADS7846 Touchscreen"
Option "Calibration" "3880 268 3936 227"
Option "SwapAxes" "0"
EndSection

Everything works as it should be. Hope it helps.