KlipperScreen/KlipperScreen #811: No logs and not working KlipperScreen on Android

110Tim706, 2022-11-25:
I tried to install Klipperscreen to use it as screen for my printer. I installed it via kiauh and it seems fine. After, that, I just copy pasted every line of this guide (https://klipperscreen.readthedocs.io/en/latest/Android/). I just get a blue picture in the app, so I want to check for the logs, but there aren't in tmp nor in logs.
[image]
[image]
[image]
[image]
[image]

alfrix, 2022-11-26:
KlipperScreen.log are on ~/printer_data/logs or /tmp/ it's explained in https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/

you are mixing methods  for android via wifi delete that script and just follow https://klipperscreen.readthedocs.io/en/latest/Android/#wifi the script should only contain the line mentioned in the docs,  

the :0 at the end of the ip is the display number not a port

110Tim706, 2022-11-26:
Thx for your reply figured it out an hour ago but this are the things I change. Maybe This will help someone else.

closed 2022-11-26 by 110Tim706