KlipperScreen/KlipperScreen #812: [BUG] Impossible to install KlipperScreen - Failed building wheel for patchelf

Guilouz, 2022-11-26:
**Describe the bug**
Since recent changes, impossible to install KlipperScreen.

**To Reproduce**
Steps to reproduce the behavior:
Just run install script

```
Collecting dbus-python
  Using cached dbus-python-1.3.2.tar.gz (605 kB)
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/.KlipperScreen-env/bin/python /home/pi/.KlipperScreen-env/lib/python3.8/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-ozqt06uu/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'meson-python>=0.8.1' 'meson>=0.60.0' ninja patchelf setuptools wheel
       cwd: None
  Complete output (184 lines):
  Collecting meson-python>=0.8.1
    Using cached meson_python-0.11.0-py3-none-any.whl (63 kB)
  Collecting meson>=0.60.0
    Using cached meson-0.64.1-py3-none-any.whl (895 kB)
  Collecting ninja
    Using cached ninja-1.11.1.tar.gz (27 kB)
    Installing build dependencies: started
    Installing build dependencies: finished with status 'done'
    Getting requirements to build wheel: started
    Getting requirements to build wheel: finished with status 'done'
      Preparing wheel metadata: started
      Preparing wheel metadata: finished with status 'done'
  Collecting patchelf
    Using cached patchelf-0.17.0.0.tar.gz (166 kB)
    Installing build dependencies: started
    Installing build dependencies: finished with status 'done'
    Getting requirements to build wheel: started
    Getting requirements to build wheel: finished with status 'done'
      Preparing wheel metadata: started
      Preparing wheel metadata: finished with status 'done'
  Collecting setuptools
    Using cached setuptools-65.6.3-py3-none-any.whl (1.2 MB)
  Collecting wheel
    Using cached wheel-0.38.4-py3-none-any.whl (36 kB)
  Collecting tomli>=1.0.0; python_version < "3.11"
    Using cached tomli-2.0.1-py3-none-any.whl (12 kB)
  Collecting pyproject-metadata>=0.5.0
    Using cached pyproject_metadata-0.6.1-py3-none-any.whl (7.4 kB)
  Collecting packaging>=19.0
    Using cached packaging-21.3-py3-none-any.whl (40 kB)
  Collecting pyparsing!=3.0.5,>=2.0.2
    Using cached pyparsing-3.0.9-py3-none-any.whl (98 kB)
  Building wheels for collected packages: ninja, patchelf
    Building wheel for ninja (PEP 517): started
    Building wheel for ninja (PEP 517): still running...
    Building wheel for ninja (PEP 517): still running...
    Building wheel for ninja (PEP 517): still running...
    Building wheel for ninja (PEP 517): still running...
    Building wheel for ninja (PEP 517): still running...
    Building wheel for ninja (PEP 517): finished with status 'done'
    Created wheel for ninja: filename=ninja-1.11.1-cp38-cp38-linux_aarch64.whl size=129291 sha256=4c919611ef61cc808c1f9843be632adb51945fc66140973d8e34f5cb2954dca0
    Stored in directory: /home/pi/.cache/pip/wheels/b1/e5/6b/270c93a0fd0063f103f3541825579bb16906970a6a1521c715
    Building wheel for patchelf (PEP 517): started
    Building wheel for patchelf (PEP 517): finished with status 'error'
    ERROR: Command errored out with exit status 1:
     command: /home/pi/.KlipperScreen-env/bin/python /tmp/tmp8eyaxfc4 build_wheel /tmp/tmpyvdk83gr
         cwd: /tmp/pip-install-_g82karj/patchelf
    Complete output (131 lines):

    --------------------------------------------------------------------------------
    -- Trying "Ninja" generator
    --------------------------------
    ---------------------------
    ----------------------
    -----------------
[... 105 lines cut ...]
      File "/tmp/pip-build-env-lzpoh4k1/overlay/lib/python3.8/site-packages/skbuild/cmaker.py", line 670, in make
        self.make_impl(clargs=clargs, config=config, source_dir=source_dir, install_target=install_target, env=env)
      File "/tmp/pip-build-env-lzpoh4k1/overlay/lib/python3.8/site-packages/skbuild/cmaker.py", line 701, in make_impl
        raise SKBuildError(

    An error occurred while building with CMake.
      Command:
        cmake --build . --target install --config Release --
      Install target:
        install
      Source directory:
        /tmp/pip-install-_g82karj/patchelf
      Working directory:
        /tmp/pip-install-_g82karj/patchelf/_skbuild/linux-aarch64-3.8/cmake-build
    Please check the install target is valid and see CMake's output for more information.
    ----------------------------------------
    ERROR: Failed building wheel for patchelf
  Successfully built ninja
  Failed to build patchelf
  ERROR: Could not build wheels for patchelf which use PEP 517 and cannot be installed directly
  ----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/.KlipperScreen-env/bin/python /home/pi/.KlipperScreen-env/lib/python3.8/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-ozqt06uu/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'meson-python>=0.8.1' 'meson>=0.60.0' ninja patchelf setuptools wheel Check the logs for full command output.
Error: pip install exited with status code 0
Unable to install dependencies, aborting install.
```

alfrix, 2022-11-26:
on a pi it will install a pre-compiled binary so it works

if i force pip to build it, then i get an error of missing `libdbus-glib-1-dev` 
so it needs to be added to the installer in the misc section

as a side note i noticed while i ran the script is that the last line was being ignored on my system, this is likely caused by the while loop, but i will remove it since that was needed for Vext and it's not used anymore

alfrix, 2022-11-26:
`sudo apt install autoconf` should resolve the issue

Guilouz, 2022-11-26:
Thanks i will try

Guilouz, 2022-11-26:
@alfrix Confirmed it's working. Thanks.

commit abb1f0d2 (2022-11-26): installer: add packages needed to build python-networkmanager
pass the req file to pip (secuential install was needed for Vext)
fixes #812

some systems don't have wheels and need compilation, the packages are needed for those cases
closed 2022-11-26 by Guilouz