KlipperScreen/KlipperScreen #813: [BUG] Layout issue when multiple printers are defined

Guilouz, 2022-11-26:
**Describe the bug**
Layout issue when multiple printers are defined

**To Reproduce**
Enable multiple printers in KlipperScreen.conf

**Expected behavior**
Layout adjustment

**Screenshots**

When multiple printer are defined : 

[image]

[image]

When only one printer is defined no issue : 

[image]

alfrix, 2022-11-26:
it doesn't seem to happen on this repo 

[image]
i even duplicated the ws message 
[image]

checked the source of your  repo, also didn't found anything that could be the cause

the only source that i could see that could be causing this is `self._screen.base_panel.content.get_allocation()`

i will change it to a screen width, it should be more stable

Guilouz, 2022-11-26:
@alfrix Could this be related to the fact that I modified the printer select panel?

To obtain this : 

[image]

Before your recent scale changes, all was working good.

Guilouz, 2022-11-26:
@alfrix Just clonned your repo, same issue.

My screen have a resolution 1024x600

[image]

[image]

alfrix, 2022-11-26:
try changing splash_screen line 46:
from: `self.labels['actions'].set_size_request(self._screen.base_panel.content.get_allocation().width, 0)`
to: `self.labels['actions'].set_size_request(self._gtk.get_content_width(), -1)`

Guilouz, 2022-11-26:
@alfrix Yes it's now ok. Thanks.

commit 20346a90 (2022-11-26): splash_screen: tweak layout sizes and remove get_allocation that can cause isssues

fixes #813

gtk: fix content_width while vertical returning 0
closed 2022-11-26 by Guilouz