KlipperScreen/KlipperScreen #814: [BUG] Display cant be woken up from set standby

dadonali, 2022-11-26:
I have the problem that the display randomly can no longer be woken from standby. 
That means, the display goes into standby after the set time.
If I then click on the display, the display stays black and you can only see the backlight. 
This can only be fixed by restarting Klipperscreen.

[image]

alfrix, 2022-11-26:
how to get the log: https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/

dadonali, 2022-11-26:
See the Log attached

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10096752/KlipperScreen.log)

alfrix, 2022-11-26:
you are  on v0.2.9-14 818f1ab the error should be fixed with c0ffa41 you need to update, sorry for the trouble

closed 2022-11-26 by alfrix

dadonali, 2022-11-26:
but v0.2.9-14 is the latest version for me? where can i get the c0ffa41 ? or you push the new version in a few days?