KlipperScreen/KlipperScreen #816: [BUG] Touch not working on MKS-Pi + MKS Pi-TS35

psjshep, 2022-11-26:
Hardware : Makerbase MKS-Pi (Rock64 architecture), Armbian 22.05.0, Linux 5.16.20-rockchip64
Screen : Makerbase MKS Pi-TS35 touchscreen, connected via ribbon cable to m/b
Software versions:
KipperScreen v0.2.9-47
Fluidd v1.21.1
klipper v0.10.0-639
moonraker v0.7.1-775

**Describe the bug**
The screen correctly displays the printer screen but none of the touch screen icons work.

I have used both xinput_calibrator from the apt install and xlibinput_calibrator (from https://github.com/kreijack/xlibinput_calibrator) & they both accept touch input for the calibration, which I have saved in the proper Xorg config location

**To Reproduce**
1. Boot MKS & wait for Klipper screen to appear
2. press any icon

**Expected behavior**
Any Icon press should result in proper navigation

**Logfile**

psjshep, 2022-11-26:
Logfile :

--------------------KlipperScreen Log Start--------------------
Git Version: v0.2.9-47-gc630178
2022-11-26 09:11:02,492 [screen.py:main()] - KlipperScreen version: v0.2.9-47-gc630178
2022-11-26 09:11:02,762 [config.py:get_config_file_location()] - Passed config (-c): /home/mks/KlipperScreen.conf
2022-11-26 09:11:02,764 [config.py:__init__()] - Config path location: /home/mks/printer_data/config/KlipperScreen.conf
2022-11-26 09:11:02,788 [config.py:log_config()] -  ===== Config File =====

=======================
2022-11-26 09:11:02,790 [config.py:__init__()] - ====== Saved Def ======

[main]
theme = material-light
screen_blanking = 900

=======================
2022-11-26 09:11:02,793 [config.py:__init__()] - Configuration validated succesfuly
2022-11-26 09:11:02,795 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
2022-11-26 09:11:02,941 [config.py:create_translations()] - Selected lang: None OS lang: en_US
2022-11-26 09:11:02,942 [config.py:install_language()] - Using system lang
2022-11-26 09:11:02,944 [config.py:install_language()] - Using lang en
2022-11-26 09:11:02,958 [screen.py:__init__()] - Screen resolution: 480x320
2022-11-26 09:11:02,961 [KlippyGtk.py:__init__()] - Font size: 11.03448275862069 (medium)
2022-11-26 09:11:03,770 [screen.py:set_screenblanking_timeout()] - Changing screen blanking to: 900
2022-11-26 09:11:03,799 [screen.py:set_screenblanking_timeout()] - Using DPMS
2022-11-26 09:11:03,845 [screen.py:initial_connection()] - Default printer: None
2022-11-26 09:11:03,846 [screen.py:connect_printer()] - Connecting to printer: Printer
2022-11-26 09:11:03,848 [screen.py:_load_panel()] - Loading panel: splash_screen
2022-11-26 09:11:03,849 [screen.py:_load_panel()] - Panel path: /home/mks/KlipperScreen/panels/splash_screen.py
2022-11-26 09:11:04,110 [screen.py:show_panel()] - Attaching panel splash_screen
2022-11-26 09:11:04,121 [screen.py:show_panel()] - Current panel hierarchy: splash_screen
2022-11-26 09:11:04,175 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-11-26 09:11:04,199 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
2022-11-26 09:11:04,200 [KlippyWebsocket.py:connect()] - Starting websocket thread
2022-11-26 09:11:04,218 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2022-11-26 09:11:04,324 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-11-26 09:11:04,359 [KlippyRest.py:send_request()] - 404 Client Error: Not Found for url: http://127.0.0.1:7125/machine/device_power/devices
2022-11-26 09:11:04,381 [KlippyRest.py:send_request()] - 404 Client Error: Not Found for url: http://127.0.0.1:7125/printer/info
2022-11-26 09:11:07,802 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-11-26 09:11:07,822 [KlippyRest.py:send_request()] - 404 Client Error: Not Found for url: http://127.0.0.1:7125/machine/device_power/devices
2022-11-26 09:11:07,840 [KlippyRest.py:send_request()] - 404 Client Error: Not Found for url: http://127.0.0.1:7125/printer/info
2022-11-26 09:11:10,814 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-11-26 09:11:10,846 [KlippyRest.py:send_request()] - 404 Client Error: Not Found for url: http://127.0.0.1:7125/machine/device_power/devices
2022-11-26 09:11:10,876 [KlippyRest.py:send_request()] - 404 Client Error: Not Found for url: http://127.0.0.1:7125/printer/info
2022-11-26 09:11:13,829 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-11-26 09:11:13,873 [KlippyRest.py:send_request()] - 404 Client Error: Not Found for url: http://127.0.0.1:7125/machine/device_power/devices
2022-11-26 09:11:13,919 [KlippyRest.py:send_request()] - 404 Client Error: Not Found for url: http://127.0.0.1:7125/printer/info
2022-11-26 09:11:16,826 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-11-26 09:11:16,871 [KlippyRest.py:send_request()] - 404 Client Error: Not Found for url: http://127.0.0.1:7125/machine/device_power/devices
2022-11-26 09:11:16,896 [KlippyRest.py:send_request()] - 404 Client Error: Not Found for url: http://127.0.0.1:7125/printer/info

alfrix, 2022-11-26:
touch related issues are a problem with the host system, not a klipperscreen bug

psjshep, 2022-11-26:
That was my 1st thought... but the calibration routines work... which displays on the display & takes input from them.... 

So if it's not KlipperScreen... what in the OS could it be?

Phil

alfrix, 2022-11-26:
there is a repo for that screen https://github.com/willngton/3DPrinterConfig/tree/main/mks_ts35

 it has a libinput.conf, maybe start there

psjshep, 2022-11-26:
Additional:
I have been messing around with xinput_calibrator, but this time using "--misclick 0" to allow any click.. & used a stylus instead of my finger...
Now with that calibration, as so long as I click "exactly" on the centre of the icon, it registers....

So.. I think the issue could be the AREA of the touch registration.... Anyone have any idea where this is?

FYI, the screen is 480x320 (3.5") screen

Phil

alfrix, 2022-11-27:
xinput_calibrator doesn't calibrate anything on most modern systems
for calibration in a system with libinput:
https://github.com/jordanruthe/KlipperScreen/issues/517#issuecomment-1069531576
there is also a for called xLibinput_calibrator but you have to compile it.

psjshep, 2022-11-27:
As I mentioned in the 1st post, I have already installed & compiled xlibinput_calibrator

I have used that output in the calibration file 99-calibration.conf but I still need to click on the exact centre of the buttons...

#517 seems to be suggesting the problem is a cheap screen - do you think this is the main problem?

Phil

alfrix, 2023-01-29:
since this is not a bug of Klipperscreen and i'm not able to help any further since it's a hw or hw config issue. i will close this

closed 2023-01-29 by alfrix

alfrix, 2024-03-15:
i found a reliable way to calibrate the screen using a tool called xtcal i detailed the process in:

https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/Touch_issues/#touch-calibration