KlipperScreen/KlipperScreen #817: [BUG] Boot Loop after Update

dadonali, 2022-11-26:
I update to the newest version, now im in a bootloop of klipperscreen and the screen doesnt load the GUI.

See only the terminal on the screen for about 2-3 sec, black screen, again the terminal.

[image]
[KlipperScreen (5).log](https://github.com/jordanruthe/KlipperScreen/files/10096915/KlipperScreen.5.log)

Aaronhrndz, 2022-11-26:
Mine shows same behaviour

dadonali, 2022-11-26:
complete new installation doesnt fix the issue

Aaronhrndz, 2022-11-26:
v0.2.9-60-ge9df355 and v0.2.9-61-ga61a74d versions shows the same behaviour

alfrix, 2022-11-26:
the log can be obtained via journalctl  if it's not dumped as explained in https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/

you can also rollback by n amount of commits to bisect where it starts working again:
```sh
cd KlipperScreen
git --reset --hard @~5
```

or do `git --reset --hard @c630178f0ca62d76195940af01c76db0d0afd7fe`

with a hash  from the commit list
https://github.com/jordanruthe/KlipperScreen/commits/master

dadonali, 2022-11-26:
Doesnt work for me :(

[image]

Aaronhrndz, 2022-11-26:
> Doesnt work for me :(
> 
> [image]

try with `git` reset --hard @c630178f0ca62d76195940af01c76db0d0afd7fe`

alfrix, 2022-11-26:
i'm sorry i wrote it wrong it's
`git reset --hard @~5`
where 5 is the number of commits/changes

dadonali, 2022-11-26:
okay git reset --hard @~5 work for me :) now lets see if the blank screen issue is fixed

Aaronhrndz, 2022-11-26:
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10097020/KlipperScreen.log)
Here's a log after restoring 5 commits ago, sadly not working. I've deleted also my custom klipper-config overrides for klipperscreen

commit 17cae917 (2022-11-26): fix bootloop with python 3.7 fixes #817
closed 2022-11-26 by alfrix