KlipperScreen/KlipperScreen #822: [FR] Running as a desktop app

lucaci32u4, 2022-11-29:
Hi.
KlipperScreen is a great interface for Pi's. But what's missing from the Klipper ecosystem is a native client (on both Linux and Windows).

I've tried running KlipperScreen on a desktop PC running a full X desktop and everything seems really close to functioning, except a few caveats:

1. Window is hardcoded to stay maximized
2. Resizing is not supported
3. Installation script can't be used, so manual installation is the way
4. No packaging for desktop installations (deb, AUR, maybe windows executable)

The following features would be nice to have:

5. Configuration does not follow the XDG standard (it should be in `~/.config` instead of `~/` to avoid polluting a user's home directory)
6. Desktop notifications for print status (at least when it finishes)
7. Upload & print directly from desktop
8. Power button should shutdown the remote machine, not the client machine

It would make sense to introduce a configuration option that enables desktop features and maybe a switch in the app. 

Packaging for deb and AUR should be pretty straightforward. Windows might be a bit problematic but I'm sure there's a way.

I'm not sure (4) would be a great idea right now because it might confuse the users as the location would be diffrent on a Pi and desktop installation, so I'm tempted to leave this idea out.

I've already poked around the code and I'm up for implementing if it fits the project's goals. Will such features be accepted into the project?

alfrix, 2022-11-30:
i have already implemented a lot of features for running it as a desktop app (i do run it as a desktop app in manjaro for while developing)

1. you can set width and height in the config i use it for development
2.  i have tried making it resizable but it's not a trivial task. you can check it in [my repo](https://github.com/alfrix/KlipperScreen/tree/resize)
3. install is meant for debian based systems, a lot of effort  would be needed to maintain more installers
4. the only type of packaging that may happen from my part is an appimage
5. config is first tried to be saved on printer_data you can relocate it on the ks folder too. [more info](https://klipperscreen.readthedocs.io/en/latest/Configuration/#klipperscreen-behaviour-towards-configuration)
6. if there is an standarized way/api between desktop enviroments to do it or if it's not too convoluted
7. usually this job is for the slicer, but it can be done
8. power button shutdown the host machine if moonraker is not connected, if it's connected it will ask printer or host where printer is the machine that has moonraker running

> Will such features be accepted into the project?

yes if it doesn't affect running ks standalone.

Dweinbach, 2023-09-13:
Hi 

regarding:

> 1. you can set width and height in the config i use it for development
> 2. i have tried making it resizable but it's not a trivial task. you can check it in [my repo](https://github.com/alfrix/KlipperScreen/tree/resize)

for #1, can you explain where/how to do this?
for #2, this link is broken, can you provide an update link?

I'm already running KS in a window I can drag, minimize, and maximize, but it always scales with the screen, and cannot be resized.

Thanks!

thijstriemstra, 2024-01-12:
> i have already implemented a lot of features for running it as a desktop app 

Nice! Looking forward to this.

alfrix, 2024-01-16:
Update on this:
1. - [x] Starts windowed when opened inside a desktop environment
2. - [x] resizing is now supported, (it could be better but works)
3. - [x] install script now asks if you want to run as service (standalone boot to ks only) if answered no, then the service is not installed and ks need to be executed manually from the .desktop shortcut
4. - [ ] to me this doesn't really affect the run as a desktop app goal, PRs welcomed
5. - [ ] this could be added without much effort preference would still be printer_data but .config can be added to the list of fallbacks
6. - [ ] possibly with [GIO](https://lazka.github.io/pgi-docs/index.html#Gio-2.0/classes/Notification.html#gio-notification) not a priority at the moment
7. - [ ] PRs welcomed
8. - [x] power let's you choose host or printer