KlipperScreen/KlipperScreen #824: [BUG] KlipperScreen Streched on 64Bit

Steven2k6, 2022-11-29:
I recently changed to 64Bit and now my KlipperScreen is streched.
I have a 480x320 RPI 3.5" Display
[config.txt](https://github.com/jordanruthe/KlipperScreen/files/10113989/config.txt)
[klippy.log](https://github.com/jordanruthe/KlipperScreen/files/1011399
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10113997/KlipperScreen.log)
5/klippy.log)

alfrix, 2022-11-30:
after a fresh install of raspbian os lite 64bit, i can confirm that your issue is something that you changed/broke
as you can see in the logs it was correct, until you changed something
[image]

Steven2k6, 2022-11-30:
It´s a bit weird, because it worked before changing to 64Bit from an 32Bit installation of an Mainsail-Image, by adding arm_64bit=1

alfrix, 2022-12-01:
closing as not a klipperscreen bug install your screen properly with the manufacturer instructions

closed 2022-12-01 by alfrix