KlipperScreen/KlipperScreen #825: [BUG] filament sensor status

CIS94500, 2022-11-29:
**Describe the bug**

the status of the filament sensor is not detected, the activation and deactivation function with the switch works

**To Reproduce**

1. Go to actions
2. Click on Extrusion

**Screenshots**

[image]

**Logfile**
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10115005/KlipperScreen.log)

Guilouz, 2022-11-29:
I can confirm this too.

alfrix, 2022-11-30:
i think this is an initial state problem, if you have the panel open it seems to work for me

does the status works if the panel is open when the filament is triggered?

Guilouz, 2022-11-30:
@alfrix yes this happens when panel is opening, if you disable and enable it again, it's ok.

it's happens since a recent PR.

CIS94500, 2022-11-30:
when the panel is open and I go from 0 to 1 or from 1 to 0, the color of the switch disappears

if I remove the filament the state does not change.

[image]
[image]

commit 4dceeb77 (2022-11-30): extrude: fix filament sensor showing status when disabled

alongside the initialization fixes this should be enough to close #825
closed 2022-11-30 by alfrix