KlipperScreen/KlipperScreen #826: Ziel?

livinhack, 2022-11-29:
https://github.com/jordanruthe/KlipperScreen/blob/68ccf5e803337a60a6d4e329f7dab1fef38108d3/ks_includes/locales/de/LC_MESSAGES/KlipperScreen.po#L252

I think "Fertig" would be a more appropriate translation!

After a Klipperscreen update, a button is currently displayed at the bottom right. "Ziel" in the German translation is more like the finish line at the end of a race. :-)  "Fertig" would fit better. Meaning like: done

MfG

alfrix, 2022-11-29:
Hi, i don't know the language
please [download the translation file](https://raw.githubusercontent.com/jordanruthe/KlipperScreen/master/ks_includes/locales/de/LC_MESSAGES/KlipperScreen.po)
edit to your liking and attach it back in this issue
i recommend using poedit

livinhack, 2022-11-30:
Ok, was the first time that I used the program. Adjusted various things.

maybe I still see room for improvement in the translation.
I'll keep an eye out for it while using it.

Renamed it to txt to attach it
[KlipperScreen.po.txt](https://github.com/jordanruthe/KlipperScreen/files/10126891/KlipperScreen.po.txt)

commit efebea2c (2022-11-30): Update German thanks to livinhack close #826
closed 2022-11-30 by alfrix