KlipperScreen/KlipperScreen #828: Layout issue with scrollbar

Guilouz, 2022-11-30:
How can I remove this unsightly scrollbar? It has appeared since recent PRs.

Before the Temp C text and 4 items fit perfectly into the frame.

Before:

[image]

After:

[image]

Guilouz, 2022-12-02:
@alfrix Can you point me what can be modified to change that ?

alfrix, 2022-12-03:
most of the spacing is done in the css so you can gtk inspector to tweak it on the fly, in my opinion scaling needs a lot more work to be more integrated and more easily tweakable

Guilouz, 2022-12-03:
Last PR fix issue. Thanks.

closed 2022-12-03 by Guilouz