KlipperScreen/KlipperScreen #834: [BUG] Can not install on OrangePi

Liechtenstein1, 2022-12-05:
I have tried installing via KIAUH and manually, i got same error on both.

 error: subprocess-exited-with-error

  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [75 lines of output]
      Collecting meson-python>=0.8.1
        Using cached meson_python-0.11.0-py3-none-any.whl (63 kB)
      Collecting meson>=0.60.0
        Using cached meson-0.64.1-py3-none-any.whl (895 kB)
      Collecting ninja
        Using cached ninja-1.11.1.tar.gz (27 kB)
        Installing build dependencies: started
        Installing build dependencies: finished with status 'done'
        Getting requirements to build wheel: started
        Getting requirements to build wheel: finished with status 'done'
        Preparing metadata (pyproject.toml): started
        Preparing metadata (pyproject.toml): finished with status 'done'
      Collecting patchelf
        Using cached patchelf-0.17.0.0.tar.gz (166 kB)
        Installing build dependencies: started
        Installing build dependencies: finished with status 'done'
        Getting requirements to build wheel: started
        Getting requirements to build wheel: finished with status 'done'
        Preparing metadata (pyproject.toml): started
        Preparing metadata (pyproject.toml): finished with status 'done'
      Collecting setuptools
        Using cached setuptools-65.6.3-py3-none-any.whl (1.2 MB)
      Collecting wheel
        Using cached wheel-0.38.4-py3-none-any.whl (36 kB)
      Collecting tomli>=1.0.0
        Using cached tomli-2.0.1-py3-none-any.whl (12 kB)
      Collecting pyproject-metadata>=0.5.0
        Using cached pyproject_metadata-0.6.1-py3-none-any.whl (7.4 kB)
      Collecting packaging>=19.0
        Using cached packaging-21.3-py3-none-any.whl (40 kB)
      Collecting pyparsing!=3.0.5,>=2.0.2
        Using cached pyparsing-3.0.9-py3-none-any.whl (98 kB)
      Building wheels for collected packages: ninja, patchelf
        Building wheel for ninja (pyproject.toml): started
        Building wheel for ninja (pyproject.toml): finished with status 'error'
        error: subprocess-exited-with-error

        × Building wheel for ninja (pyproject.toml) did not run successfully.
        │ exit code: 1
        ╰─> [9 lines of output]
            Traceback (most recent call last):
              File "/tmp/pip-build-env-foegwaf1/overlay/lib/python3.10/site-packages/skbuild/setuptools_wrap.py", line 612, in setup
                cmkr = cmaker.CMaker(cmake_executable)
              File "/tmp/pip-build-env-foegwaf1/overlay/lib/python3.10/site-packages/skbuild/cmaker.py", line 148, in __init__
                self.cmake_version = get_cmake_version(self.cmake_executable)
              File "/tmp/pip-build-env-foegwaf1/overlay/lib/python3.10/site-packages/skbuild/cmaker.py", line 103, in get_cmake_version
                raise SKBuildError(

            Problem with the CMake installation, aborting build. CMake executable is cmake
            [end of output]

        note: This error originates from a subprocess, and is likely not a problem with pip.
        ERROR: Failed building wheel for ninja
        Building wheel for patchelf (pyproject.toml): started
        Building wheel for patchelf (pyproject.toml): finished with status 'error'
        error: subprocess-exited-with-error

        × Building wheel for patchelf (pyproject.toml) did not run successfully.
        │ exit code: 1
        ╰─> [9 lines of output]
            Traceback (most recent call last):
              File "/tmp/pip-build-env-w5u0ttnt/overlay/lib/python3.10/site-packages/skbuild/setuptools_wrap.py", line 612, in setup
                cmkr = cmaker.CMaker(cmake_executable)
              File "/tmp/pip-build-env-w5u0ttnt/overlay/lib/python3.10/site-packages/skbuild/cmaker.py", line 148, in __init__
                self.cmake_version = get_cmake_version(self.cmake_executable)
              File "/tmp/pip-build-env-w5u0ttnt/overlay/lib/python3.10/site-packages/skbuild/cmaker.py", line 103, in get_cmake_version
                raise SKBuildError(

            Problem with the CMake installation, aborting build. CMake executable is cmake
            [end of output]

        note: This error originates from a subprocess, and is likely not a problem with pip.
        ERROR: Failed building wheel for patchelf
      Failed to build ninja patchelf
      ERROR: Could not build wheels for ninja, patchelf, which is required to install pyproject.toml-based projects
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
Error: pip install exited with status code 0
Unable to install dependencies, aborting install.

alfrix, 2022-12-05:
install cmake or build-essentials in your platform

closed 2022-12-05 by alfrix

Liechtenstein1, 2022-12-06:
It worked, thank you very much!