KlipperScreen/KlipperScreen #837: [BUG] Recent Get_Probe changes lead to probe trigger ignored and, crash.

danorder, 2022-12-11:
**Describe the bug**
A clear and concise description of what the bug is.

It would appear the commits involving the new get_probe  function lead to a crash under Delta when using zcalibrate button. 

**To Reproduce**
run  delta zcalibrate probe trigger is ignored and crashes. 
Run probe_calibrate from ui terminal it runs as expected. probe trigger is not ignored.   

The only error reporting in log pertaining to this. 

2022-12-10 16:14:14,406 [screen.py:wake_screen()] - Screen wake up
2022-12-10 16:14:14,435 [screen.py:_load_panel()] - Loading panel: job_status
2022-12-10 16:14:14,436 [screen.py:_load_panel()] - Panel path: /home/pi/KlipperScreen/panels/job_status.py
2022-12-10 16:14:14,595 [screen.py:attach_panel()] - Current panel hierarchy: job_status
2022-12-10 16:14:14,596 [job_status.py:set_state()] - Changing job_status state from 'standby' to 'printing'
2022-12-10 16:14:14,714 [job_status.py:process_update()] - Changing filename: 'None' to 'VORON_CUBE_20mm_V3.gcode'
2022-12-10 16:14:14,715 [job_status.py:update_file_metadata()] - Update Metadata. File: VORON_CUBE_20mm_V3.gcode Size: 964498
2022-12-10 16:19:14,691 [screen.py:show_screensaver()] - Showing Screensaver
2022-12-10 16:19:29,560 [screen.py:close_screensaver()] - Closing Screensaver
2022-12-10 16:19:29,562 [screen.py:wake_screen()] - Screen wake up
2022-12-10 16:20:20,465 [screen.py:_load_panel()] - Loading panel: fine_tune
2022-12-10 16:20:20,466 [screen.py:_load_panel()] - Panel path: /home/pi/KlipperScreen/panels/fine_tune.py
2022-12-10 16:20:20,541 [screen.py:attach_panel()] - Current panel hierarchy: job_status > fine_tune
2022-12-10 16:20:35,281 [screen.py:_menu_go_back()] - #### Menu go back
2022-12-10 16:20:35,322 [screen.py:attach_panel()] - Current panel hierarchy: job_status
2022-12-10 16:20:36,320 [KlippyGtk.py:Dialog()] - Showing dialog <Gtk.Dialog object at 0xeac7fc88 (GtkDialog at 0x1af1378)>
2022-12-10 16:20:36,929 [KlippyGtk.py:remove_dialog()] - Removing Dialog
2022-12-10 16:20:36,929 [job_status.py:cancel_confirm()] - Canceling print
2022-12-10 16:20:36,930 [job_status.py:set_state()] - Changing job_status state from 'printing' to 'cancelling'
2022-12-10 16:20:36,936 [printer.py:get_probe()] - Probe type: probe
2022-12-10 16:20:36,942 [KlippyWebsocket.py:print_cancel()] - Sending printer.print.cancel
2022-12-10 16:20:37,304 [job_status.py:set_state()] - Changing job_status state from 'cancelling' to 'cancelled'
2022-12-10 16:20:37,306 [printer.py:get_probe()] - Probe type: probe
2022-12-10 16:20:52,104 [KlippyWebsocket.py:print_start()] - Sending printer.print.start
2022-12-10 16:20:52,631 [job_status.py:set_state()] - Changing job_status state from 'cancelled' to 'printing'

alfrix, 2022-12-11:
The snippet you have cut doesn't make sense for the issue described, provide the full log.

the description provided is not helpful, as in there is no mention on what kind of probe is used or what method is selected for zcalibrate since delta should have at least 3 different methods with a probe

danorder, 2022-12-11:
The  probe is just labeled as probe in the config. but further the current
solution I have listed on the delta support page is to revert back to tag
0.2.8 . And sorry the method is Probe_calibrate or the button in klipper
screen is named "probe" under zcalibrate.
The type is just a push button or "kossel"  style. hopefully this log is of
use. Im not certain klipper screen is catching it. but i can also run a few
things if needed.

[probe]
pin: !PA11
x_offset: 0
y_offset: 0
#z_offset: 16.00 #the distance between nozzle and level switch
speed: 2

alfrix, 2022-12-11:
> hopefully this log is of use.

the log is still missing, and of course the log must contain the moment of the problem occurred

danorder, 2022-12-11:
> > hopefully this log is of use.
> 
> the log is still missing, and of course the log must contain the moment of the problem occurred

Sorry for some reason gmail didn't upload it . I should have responded on here. 

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10203451/KlipperScreen.log)

alfrix, 2022-12-12:
you never opened the zcalibrate panel in that log as i said:

>  the log must contain the moment of the problem occurred

danorder, 2022-12-12:
I'll delete the logs and run this again.  and, get back to you shortly.  i figured the mention was missing which is why i looked  at recent commits first making this thread.

danorder, 2022-12-12:
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10204509/KlipperScreen.log)
 
that seemed to work unknown command testz

alfrix, 2022-12-12:
no, it has nothing to do with the initial testz, it's this:
```
2022-12-11 20:11:50,908 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: G0 Z3 F120.0
```

alfrix, 2022-12-12:
thank you for taking the time to report and log the issue

closed 2022-12-12 by alfrix

danorder, 2022-12-12:
Ah k i thought it was because it didn't recognize that and, went straight to a move. Im  clearly rushing around to much.  This is a pretty important feature for beginners other then being easy.  Also thank you for your time and implementing it when i requested delta support some time back.

danorder, 2022-12-12:
zcalibrate probe is mostly fixed. however the screen freezes after Hitting accept. 

2022-12-11 22:57:45,867 [screen.py:attach_panel()] - Current panel hierarchy: main_panel > config > zcalibrate
2022-12-11 22:57:45,867 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=+0.001
2022-12-11 22:57:45,914 [zcalibrate.py:process_update()] - // unknown command:"testz"
2022-12-11 22:57:48,231 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: G91
G0 Z3 F120.0
2022-12-11 22:57:48,233 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: G90
2022-12-11 22:57:48,235 [zcalibrate.py:_move_to_position()] - Detected delta kinematics calibrating at 0,0
2022-12-11 22:57:48,236 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: G0 X0 Y0 F3000
2022-12-11 22:57:48,237 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: PROBE_CALIBRATE
2022-12-11 22:58:44,733 [zcalibrate.py:change_distance()] - ### Distance 5
2022-12-11 22:58:49,718 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-5
2022-12-11 22:58:52,460 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-5
2022-12-11 22:58:54,486 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-5
2022-12-11 22:58:59,555 [zcalibrate.py:change_distance()] - ### Distance 1
2022-12-11 22:59:00,504 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-11 22:59:02,157 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-11 22:59:03,801 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-11 22:59:07,981 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-11 22:59:10,501 [zcalibrate.py:change_distance()] - ### Distance .1
2022-12-11 22:59:11,042 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.1
2022-12-11 22:59:12,694 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.1
2022-12-11 22:59:14,383 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.1
2022-12-11 22:59:15,973 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.1
2022-12-11 22:59:18,598 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.1
2022-12-11 22:59:20,582 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.1
2022-12-11 22:59:25,285 [zcalibrate.py:change_distance()] - ### Distance 1
2022-12-11 22:59:26,089 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-11 22:59:28,624 [zcalibrate.py:change_distance()] - ### Distance .1
2022-12-11 22:59:29,571 [zcalibrate.py:change_distance()] - ### Distance .5
2022-12-11 22:59:30,347 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.5
2022-12-11 22:59:31,849 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.5
2022-12-11 22:59:35,692 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.5
2022-12-11 22:59:39,739 [zcalibrate.py:change_distance()] - ### Distance .05
2022-12-11 22:59:40,224 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.05
2022-12-11 22:59:43,161 [zcalibrate.py:change_distance()] - ### Distance .01
2022-12-11 22:59:44,320 [zcalibrate.py:change_distance()] - ### Distance .05
2022-12-11 22:59:44,803 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.05
2022-12-11 22:59:48,583 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.05
2022-12-11 22:59:51,220 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.05
2022-12-11 22:59:54,277 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.05
2022-12-11 22:59:56,189 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.05
2022-12-11 22:59:57,816 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.05
2022-12-11 22:59:59,933 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.05
2022-12-11 23:00:02,090 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.05
2022-12-11 23:00:03,796 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.05
2022-12-11 23:00:05,793 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.05
2022-12-11 23:00:07,732 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.05
2022-12-11 23:00:09,818 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.05
2022-12-11 23:00:17,134 [zcalibrate.py:change_distance()] - ### Distance .01
2022-12-11 23:00:17,704 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.01
2022-12-11 23:00:20,299 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.01
2022-12-11 23:00:23,114 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.01
2022-12-11 23:00:25,506 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.01
2022-12-11 23:00:31,613 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.01
2022-12-11 23:00:38,431 [zcalibrate.py:accept()] - Accepting Z position
2022-12-11 23:00:38,432 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: ACCEPT
2022-12-11 23:00:38,491 [KlippyGtk.py:Dialog()] - Showing dialog <Gtk.Dialog object at 0xeb97b0a8 (GtkDialog at 0x19e61a0)>

danorder, 2022-12-12:
Update a reboot seems to have solved it i'll run a few more times. but other then a initial zhop it seems to work now.

commit 82b8cbcc (2022-12-12): zcalibrate: fix initial zhop causing issues for some machines fixes #837