KlipperScreen/KlipperScreen #844: [BUG] Klipper screen freezes after Zcalibrate probe accept ,

danorder, 2022-12-13:
Follow up. Probe no longer crashes  However hitting accept the screen freezes. 

2022-12-12 18:28:34,348 [screen.py:attach_panel()] - Current panel hierarchy: main_panel
2022-12-12 18:28:35,261 [screen.py:_go_to_submenu()] - #### Go to submenu config
2022-12-12 18:28:35,262 [screen.py:_go_to_submenu()] - #### Menu __main
2022-12-12 18:28:35,300 [screen.py:attach_panel()] - Current panel hierarchy: main_panel > config
2022-12-12 18:28:35,308 [menu.py:arrangeMenuItems()] - X > config bedlevel
2022-12-12 18:28:35,332 [menu.py:arrangeMenuItems()] - X > config retraction
2022-12-12 18:28:36,985 [screen.py:attach_panel()] - Current panel hierarchy: main_panel > config > zcalibrate
2022-12-12 18:28:36,987 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=+0.001
2022-12-12 18:28:37,029 [zcalibrate.py:process_update()] - // unknown command:"testz"
2022-12-12 18:28:41,515 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: G91
G0 Z3 F120.0
2022-12-12 18:28:41,519 [zcalibrate.py:_move_to_position()] - Detected delta kinematics calibrating at 0,0
2022-12-12 18:28:41,519 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: G0 X0 Y0 F3000
2022-12-12 18:28:41,521 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: PROBE_CALIBRATE
2022-12-12 18:29:26,196 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:29:28,471 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:29:30,320 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:29:32,124 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:29:33,870 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:29:35,410 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:29:36,805 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:29:38,109 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:29:40,691 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:29:42,671 [zcalibrate.py:change_distance()] - ### Distance 5
2022-12-12 18:29:43,655 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-5
2022-12-12 18:29:45,856 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-5
2022-12-12 18:29:59,962 [zcalibrate.py:change_distance()] - ### Distance 1
2022-12-12 18:30:01,637 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:30:03,662 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:30:08,631 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:30:10,727 [zcalibrate.py:change_distance()] - ### Distance .5
2022-12-12 18:30:13,471 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.5
2022-12-12 18:30:16,390 [zcalibrate.py:change_distance()] - ### Distance .01
2022-12-12 18:30:16,876 [zcalibrate.py:change_distance()] - ### Distance .05
2022-12-12 18:30:17,495 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.05
2022-12-12 18:30:19,562 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.05
2022-12-12 18:30:20,999 [zcalibrate.py:change_distance()] - ### Distance .01
2022-12-12 18:30:22,747 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.01
2022-12-12 18:30:25,529 [zcalibrate.py:accept()] - Accepting Z position
2022-12-12 18:30:25,531 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: ACCEPT
2022-12-12 18:30:25,606 [KlippyGtk.py:Dialog()] - Showing dialog <Gtk.Dialog object at 0xebcc5448 (GtkDialog at 0x2078360)>

#837

alfrix, 2022-12-13:
everything seems fine in the snippet a confirmation dialog should be shown asking to save

danorder, 2022-12-13:
ive run this a few time. i  keep getting the same log snippets then it hangs. The only way i can get around it is saving from Mainsail or fluidd afterwards. Is there another way to generate a better log?

alfrix, 2022-12-13:
the save dialog seems to be failing to show, but still locking, i am not sure how that would happen, i may add a couple of extra show commands that may help

danorder, 2022-12-13:
checked panel comes up however save_config doesnt appear to run. ill see if it works after a reboot 
log shows no further lines after 2022-12-13 15:25:53,115 [KlippyGtk.py:Dialog()] - Showing dialog <Gtk.Dialog object at 0xeb32f808 (GtkDialog at 0x29161b0)>

2022-12-13 15:23:26,181 [screen.py:attach_panel()] - Current panel hierarchy: main_panel > config
2022-12-13 15:23:26,186 [menu.py:arrangeMenuItems()] - X > config bedlevel
2022-12-13 15:23:26,204 [menu.py:arrangeMenuItems()] - X > config retraction
2022-12-13 15:23:27,036 [screen.py:_load_panel()] - Loading panel: zcalibrate
2022-12-13 15:23:27,036 [screen.py:_load_panel()] - Panel path: /home/pi/KlipperScreen/panels/zcalibrate.py
2022-12-13 15:23:27,046 [printer.py:get_probe()] - Probe type: probe
2022-12-13 15:23:27,046 [zcalibrate.py:__init__()] - Z offset: 17.71
2022-12-13 15:23:27,102 [zcalibrate.py:__init__()] - Available functions for calibration: ['probe', 'delta', 'delta_manual']
2022-12-13 15:23:27,118 [screen.py:attach_panel()] - Current panel hierarchy: main_panel > config > zcalibrate
2022-12-13 15:23:27,119 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=+0.001
2022-12-13 15:23:27,162 [zcalibrate.py:process_update()] - // unknown command:"testz"
2022-12-13 15:23:28,699 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: G28
2022-12-13 15:23:28,702 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: G91
G0 Z3 F120.0
2022-12-13 15:23:28,705 [zcalibrate.py:_move_to_position()] - Detected delta kinematics calibrating at 0,0
2022-12-13 15:23:28,706 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: G0 X0 Y0 F3000
2022-12-13 15:23:28,710 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: PROBE_CALIBRATE
2022-12-13 15:23:35,094 [zcalibrate.py:process_update()] - !! move out of range: 0.328 -0.411 422.969 [48041.689]
2022-12-13 15:25:50,306 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-13 15:25:53,022 [zcalibrate.py:accept()] - Accepting Z position
2022-12-13 15:25:53,023 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: ACCEPT
2022-12-13 15:25:53,115 [KlippyGtk.py:Dialog()] - Showing dialog <Gtk.Dialog object at 0xeb32f808 (GtkDialog at 0x29161b0)>

danorder, 2022-12-13:
Update rebooting doesn't solve. However. it appears to prompt is delayed for some reason.  

current behavior run zcalibrate probe it probes lower it etc accept, save prompt shows up hit save it disappears quite quick.  does nothing wait awhile it shows up again,  hit save it  , saves then reboot applying changes. 

new log 

--------------------KlipperScreen Log Start--------------------
Git Version: v0.3.1-9-gb96adf5
2022-12-13 15:34:16,530 [screen.py:main()] - KlipperScreen version: v0.3.1-9-gb96adf5
2022-12-13 15:34:16,739 [config.py:get_config_file_location()] - Passed config (-c): /home/pi/KlipperScreen.conf
2022-12-13 15:34:16,740 [config.py:__init__()] - Config path location: /home/pi/printer_data/config/KlipperScreen.conf
2022-12-13 15:34:16,752 [config.py:log_config()] -  ===== Config File =====

=======================
2022-12-13 15:34:16,754 [config.py:__init__()] - ====== Saved Def ======

[main]
screen_blanking = 300
use_dpms = False
print_sort_dir = date_desc

=======================
2022-12-13 15:34:16,756 [config.py:__init__()] - Configuration validated succesfuly
2022-12-13 15:34:16,756 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
2022-12-13 15:34:16,847 [config.py:create_translations()] - Selected lang: None OS lang: en_US
2022-12-13 15:34:16,848 [config.py:install_language()] - Using system lang
2022-12-13 15:34:16,849 [config.py:install_language()] - Using lang en
2022-12-13 15:34:16,857 [screen.py:__init__()] - Screen resolution: 1024x600
2022-12-13 15:34:16,858 [KlippyGtk.py:__init__()] - Font size: 20.7 (medium)
2022-12-13 15:34:17,500 [screen.py:set_screenblanking_timeout()] - Changing screen blanking to: 300
2022-12-13 15:34:17,500 [screen.py:set_screenblanking_timeout()] - Not using DPMS
2022-12-13 15:34:17,509 [screen.py:initial_connection()] - Default printer: None
2022-12-13 15:34:17,510 [screen.py:connect_printer()] - Connecting to printer: Printer
2022-12-13 15:34:17,510 [screen.py:_load_panel()] - Loading panel: splash_screen
2022-12-13 15:34:17,510 [screen.py:_load_panel()] - Panel path: /home/pi/KlipperScreen/panels/splash_screen.py
2022-12-13 15:34:17,570 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen
2022-12-13 15:34:17,592 [KlippyRest.py:send_request()] - host=127.0.0.1, port=7125>>Max retries exceeded with url>>Failed to establish a new connection>>[Errno 111] Connection refused
2022-12-13 15:34:27,375 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
2022-12-13 15:34:27,395 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-12-13 15:34:27,409 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
2022-12-13 15:34:27,410 [KlippyWebsocket.py:connect()] - Starting websocket thread
2022-12-13 15:34:27,420 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2022-12-13 15:34:27,421 [_logging.py:warning()] - websocket connected
2022-12-13 15:34:27,436 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-12-13 15:34:27,461 [KlippyRest.py:send_request()] - 404 Client Error>>Not Found for url>>http://127.0.0.1:7125/machine/device_power/devices
2022-12-13 15:34:27,481 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/info
2022-12-13 15:34:27,730 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/objects/query?configfile
2022-12-13 15:34:27,734 [printer.py:reinit()] - Klipper version: v0.11.0-13-g201f715b
2022-12-13 15:34:27,735 [printer.py:reinit()] - # Extruders: 1
2022-12-13 15:34:27,735 [printer.py:reinit()] - # Temperature devices: 3
2022-12-13 15:34:27,735 [printer.py:reinit()] - # Fans: 2
2022-12-13 15:34:27,735 [printer.py:reinit()] - # Output pins: 2
2022-12-13 15:34:27,736 [KlippyWebsocket.py:object_subscription()] - Sending printer.objects.subscribe
2022-12-13 15:34:28,017 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/objects/query?bed_mesh&configfile&display_status&extruder&fan&gcode_move&heater_bed&idle_timeout&pause_resume&print_stats&toolhead&virtual_sdcard&webhooks&motion_report&firmware_retraction&exclude_object&extruder&heater_bed&temperature_sensor Armv7 mcu&temperature_sensor mcu_temp&fan&heater_fan heat_sink_fan&filament_switch_sensor filament_sensor&output_pin LED_pin&output_pin LED_logo
2022-12-13 15:34:28,022 [printer.py:change_state()] - Changing state from 'disconnected' to 'ready'
2022-12-13 15:34:28,046 [printer.py:change_state()] - Adding callback for state: ready
2022-12-13 15:34:28,096 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/temperature_store
2022-12-13 15:34:28,122 [printer.py:init_temp_store()] - Temp store: ['heater_bed', 'temperature_sensor Armv7 mcu', 'temperature_sensor mcu_temp', 'extruder']
2022-12-13 15:34:28,145 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/config
2022-12-13 15:34:28,146 [screen.py:init_tempstore()] - Temperature store size: 1200
2022-12-13 15:34:28,147 [files.py:initialize()] - Gcodes path: /home/pi/printer_1_data/gcodes
2022-12-13 15:34:28,147 [KlippyWebsocket.py:get_file_list()] - Sending server.files.list
[... 116 lines cut ...]
2022-12-13 15:46:47,886 [printer.py:reinit()] - # Extruders: 1
[... cut ...]
2022-12-13 15:46:48,422 [screen.py:attach_panel()] - Current panel hierarchy: main_panel
2022-12-13 15:46:50,543 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/temperature_store
2022-12-13 15:46:50,560 [printer.py:init_temp_store()] - Temp store: ['heater_bed', 'temperature_sensor Armv7 mcu', 'temperature_sensor mcu_temp', 'extruder']
2022-12-13 15:46:50,574 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/config
2022-12-13 15:46:50,575 [screen.py:init_tempstore()] - Temperature store size: 1200

Bahoue, 2022-12-18:
Hello, I have the same problem, very fast tooltip then freeze. Here is a video showing the problem: https://www.youtube.com/watch?v=Vsns2dQ5srw

northcoastam, 2022-12-22:
Not sure if this is relevant but here goes. I had a separate cfg file for the probe related stuff, including the z_offset.  I experienced a hang after choosing probe accept a couple of times.  Then I moved the [probe] z_offset entry to the main printer.cfg file and was able to accept the calibrate value without hanging.  Maybe it was a coincidence, please delete if this is irrelevant.

Bahoue, 2022-12-22:
> Not sure if this is relevant but here goes. I had a separate cfg file for the probe related stuff, including the z_offset. I experienced a hang after choosing probe accept a couple of times. Then I [...]

I did the test and it did not change the problem.

Guilouz, 2023-01-03:
You can fix this by replacing 'accept' definition in zcalibrate.py file by this :

```
def accept(self, widget):
        logging.info("Accepting Z position")
        self._screen._ws.klippy.gcode_script("ACCEPT")
```

It's now working for me.

erax0r, 2023-01-04:
i have been having this issue myself. thanks for posting a solution, I will try it out tomorrow after my print finishes!

northcoastam, 2023-01-04:
Don't get your hopes up :P
It didn't solve the problem for Bahoué.

Guilouz, 2023-01-07:
@Bahoue Another workaround is to replace this line in screen.py file : 
```
if "SAVE_CONFIG" in data and self.printer.state == "ready:":
```
by this one:
```
if "SAVE_CONFIG" in data and self.printer.state == "ready" and not "probe: z_offset:" in data:
```

This makes it possible to accept the validation of the Z Offset without the request to save (which freezes the screen). Save config will then have to be done manually via the Configurations menu.

danorder, 2023-01-12:
sighhhhhh the solution is so simple. accept it false crashes. save_config from fluidd/mainsail.  now users don't need to mess with files or other pointless issues later once alfrix is back from presumed break.

alfrix, 2023-03-10:
[a6bb0fd](https://github.com/jordanruthe/KlipperScreen/commit/a6bb0fd02e79f8a12918ef838e83385060b3bb02) should have fixed this, i'll reopen if it's not the case

closed 2023-03-10 by alfrix