KlipperScreen/KlipperScreen #845: [BUG] Minor. initial  testz out of range error

danorder, 2022-12-13:
Printer delta. 

Pushing zcalibrate probe leads to move out of range warning however proceeds to run the calibration with expected behavior. 

2022-12-12 18:28:34,348 [screen.py:attach_panel()] - Current panel hierarchy: main_panel
2022-12-12 18:28:35,261 [screen.py:_go_to_submenu()] - #### Go to submenu config
2022-12-12 18:28:35,262 [screen.py:_go_to_submenu()] - #### Menu __main
2022-12-12 18:28:35,300 [screen.py:attach_panel()] - Current panel hierarchy: main_panel > config
2022-12-12 18:28:35,308 [menu.py:arrangeMenuItems()] - X > config bedlevel
2022-12-12 18:28:35,332 [menu.py:arrangeMenuItems()] - X > config retraction
2022-12-12 18:28:36,985 [screen.py:attach_panel()] - Current panel hierarchy: main_panel > config > zcalibrate
2022-12-12 18:28:36,987 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=+0.001
2022-12-12 18:28:37,029 [zcalibrate.py:process_update()] - // unknown command:"testz"
2022-12-12 18:28:41,515 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: G91
G0 Z3 F120.0
2022-12-12 18:28:41,519 [zcalibrate.py:_move_to_position()] - Detected delta kinematics calibrating at 0,0
2022-12-12 18:28:41,519 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: G0 X0 Y0 F3000
2022-12-12 18:28:41,521 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: PROBE_CALIBRATE
2022-12-12 18:29:26,196 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:29:28,471 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:29:30,320 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:29:32,124 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:29:33,870 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:29:35,410 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:29:36,805 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:29:38,109 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:29:40,691 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:29:42,671 [zcalibrate.py:change_distance()] - ### Distance 5
2022-12-12 18:29:43,655 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-5
2022-12-12 18:29:45,856 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-5
2022-12-12 18:29:59,962 [zcalibrate.py:change_distance()] - ### Distance 1
2022-12-12 18:30:01,637 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:30:03,662 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:30:08,631 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-1
2022-12-12 18:30:10,727 [zcalibrate.py:change_distance()] - ### Distance .5
2022-12-12 18:30:13,471 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.5
2022-12-12 18:30:16,390 [zcalibrate.py:change_distance()] - ### Distance .01
2022-12-12 18:30:16,876 [zcalibrate.py:change_distance()] - ### Distance .05
2022-12-12 18:30:17,495 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.05
2022-12-12 18:30:19,562 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.05
2022-12-12 18:30:20,999 [zcalibrate.py:change_distance()] - ### Distance .01
2022-12-12 18:30:22,747 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: TESTZ Z=-.01
2022-12-12 18:30:25,529 [zcalibrate.py:accept()] - Accepting Z position
2022-12-12 18:30:25,531 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: ACCEPT
2022-12-12 18:30:25,606 [KlippyGtk.py:Dialog()] - Showing dialog <Gtk.Dialog object at 0xebcc5448 (GtkDialog at 0x2078360)>

#837

alfrix, 2022-12-13:
that is by design because there is no status that says it's calibrating and accepting testz

closed 2022-12-13 by alfrix

danorder, 2022-12-13:
i'll make a mention in forum to ignore this. or its intended rather. then
it would just be the freezing issue in the other commit. Everything else on
delta should in theory be fine then.