KlipperScreen/KlipperScreen #850: [BUG] Highlight bars of menu items are cut off

DrPacemaker, 2022-12-15:
**Describe the bug**
I have just updated KlipperScreen to V0.3.1-9 and noticed the bottom highlighters in main_menu items are missing. As you can see in the screenshots the 'Actions', 'Temperature', 'Homing' and 'Configuration' has a stripped bottom line and the colored highlight bars of the menu items are missing/not visible. My guess is this is somehow related to this commit: [7120c424d4aa24ab63158faef9989ca80aa8704a](https://github.com/jordanruthe/KlipperScreen/commit/7120c424d4aa24ab63158faef9989ca80aa8704a)

Are you experiencing the same? I have this issue on XSDL emulation and on my 5" screen (Waveshare).

**Screenshots**
[image]
[image]
[image]

alfrix, 2023-02-12:
need the log to check resolution, and font size when this happens

alfrix, 2023-03-22:
closing since no response has been given, there is no UI cut at the jpg resolution at any font-size