KlipperScreen/KlipperScreen #851: KlipperScreen one host multiple printers and tablets

solo2424, 2022-12-16:
So I have been trying to get KlipperScreen installed on one linux mint host with two printers and two separate Galaxy tablets. I believe I read somewhere that you need two instants of KlipperScreen installed so I did the following.  

I navigated to my RedRover_data (printer1) directory. and installed KlipperScreen. I was able to get it all set up and working.

I then navigated to the WhiteLighting_data (printer2) directory and started a second install of KlipperScreen. As soon as that install started, my RedRover instants stopped running. Now neither are working. Can anyone help guide me on this set up? I even tried creating a KlipperScreen.conf file and placing it at the root of my username folder, at the root of the KlipperScreen folder of each install and in the config folder of each printer folder.

KlipperScreen.conf example
[printer RedRover] moonraker_host: 127.0.0.1 moonraker_port: 7125

[printer WhiteLighting] moonraker_host: 127.0.0.1 moonraker_port: 7126

solo2424, 2022-12-16:
Forgot to add...I also edited the "launch_KlipperScreen.sh" in each script directory with DISPLAY=192.168.0.7:0 $KS_XCLIENT for tablet 1 and DISPLAY=192.168.0.89:1 $KS_XCLIENT for tablet 2 in accordance with my xserver installs to have it work over wifi.

alfrix, 2022-12-17:
don't install ks more then once, just open 2 instances using a launch script like this:
`DISPLAY=192.168.0.7:0 $KS_XCLIENT && DISPLAY=192.168.0.89:0 $KS_XCLIENT`

and don't forget that every configuration in klipperscreen.conf should be in a new line, as shown in the docs.

solo2424, 2022-12-17:
Thanks for the advise, I made some progress. I now have my 0.7 tablet giving me the option to choose what printer to display, but the 0.89 tablet is not launching klipperscreen. just to clarify with your example code. I had to change my second display to "DISPLAY=192.168.0.89:1 $KS_XCLIENT" because that is what is displayed on the second tablet.  So attached is what I have in the KlipperScreen.conf file.

[image]

solo2424, 2022-12-17:
After further investigating, I think the issue may be with my conf file. Regardless if I select RedRover or WhiteLighitng, the KlipperScreen only controls RedRover. Here is a screenshot of my KlipperScreen.conf. I have it stored at the root of my user home folder. 
[image]

solo2424, 2022-12-27:
Anyone have any thoughts on this?

solo2424, 2022-12-29:
Ok, so I have made some progress. I am able to access two printers connected to the same host from one tablet running KlipperScreen using the using the attached KlipperScreen.conf file, which I placed on the root of the user folder "~/home/<user name>/"
[image]

But I still have not found a way to connect to two separate tablets at the same time. I believe this issue is caused by my "launch_KlipperScreen.sh" file. This is where I was told to add the display line below. 

DISPLAY=192.168.0.7:0 $KS_XCLIENT && DISPLAY=192.168.0.89:1 $KS_XCLIENT

The first IP address is the only tablet that communicates with KlipperScreen. So if I put 0.7 first then the tablet with ip address ending in 0.7 connects with KS. if I use 0.89 first, then the tablet with the ip address ending in 0.89 connects to KS. But I can not get them both to connect at the same time.

solo2424, 2022-12-29:
I was able to resolve my issue. So to sum up, in order to get multiple printers on the same host, running KlipperScreen on multiple tablets/phones, you need to first edit the KlipperScreen.conf file that is stored "~/home/<username>/ See a sample of mine below.

[KlipperScreen.conf ](https://pastebin.com/gPJan9ZJ )
 

Second, you need to edit the "launch_KlipperScreen.sh" to launch on the proper displays. I just followed this tutorial.[ https://klipperscreen.readthedocs.io/en/latest/Android/](https://klipperscreen.readthedocs.io/en/latest/Android/) 

Where I deviated, is to accommodate for multiple tablets. So instead of following the tutorial, I changed the "DISPLAY" line to account for multiple tablets running xserver by adding an "&" between the displays. See example below. 

DISPLAY=192.168.0.7:0 $KS_XCLIENT & DISPLAY=192.168.0.89:0 $KS_XCLIENT 

[Sample-launch_KlipperScreen.sh](https://pastebin.com/aPc3Z8ms) 

I hope this helps anyone trying a similar set up. Feel free to reach out if you have any questions.

closed 2022-12-29 by solo2424

lucasavendano, 2023-04-10:
Hi @solo2424, i did exaclty what you did, the only difference is 3 printers instead 2 (linux mint server+3 printers+1 instance of klipperscreen controlling 3 printers via wi fi with an old tablet)

First used two displays tablet+my cell phone, but as i have a very low end server an realized klipperscreen takes a lot paring displays i decided to use only my old tablet but, here is my question in case you had or have the same "issue":

When not monitoring or controlling any printer i lock screen in my tablet for saving battery (not always powered), after a certain time, when i need to unlock tablet, klipperscreen is like "freezed", not responding, the only way to pass this is closing xsdl server app and restarting --during this process there is a heavy load on server i want to avoid specially when printing but seems there is no big risk. 

The real thing, if the same thing happend to you, is if you figure out a way to unlock tablet after a while and finding klipperscreen alive.

I did: disabled screensaver in server (headless) and in xsdl also, but with no success.

Before having tablet display always on and always powered i wonder if there is a way to solve this.

This issue on github is closed, i don´t know if you will read this.