KlipperScreen/KlipperScreen #852: [BUG] Installation on Bullseye with latest MainsailOS 0.71 required your "last resort"

JohnOCFII, 2022-12-19:
**Describe the bug**
Just FYI -- I did a clean install of the new MainsailOS 0.71 64bit Bullseye on a Raspberry Pi 3B+ today.  It required your "last resort" of `sudo bash -c "echo needs_root_rights=yes>>/etc/X11/Xwrapper.config"` to get KlipperScreen to start and display properly.

**To Reproduce**
Follow your manual installation procedure as shown here: https://klipperscreen.readthedocs.io/en/latest/Installation/#auto-install

Then follow your troubleshooting procedure as shown here: https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/

Follow all steps (results were as expected) and then finally try the last resort -- KlipperScreen started up immediately after that.

**Expected behavior**
Expected not to need the troubleshooting steps or the manual update to Xwrapper.config

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logfile**
Attach your [klipperscreen.log](https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/)

Attached -- but this was after installation was fixed as discussed above.
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10255283/KlipperScreen.log)

ReXT3D, 2022-12-20:
Coincidentally the 32 bit version of Bullseye exibits the exact same behaviour with Raspberry Pi 3B+ but not with Raspberry Pi 4B.  The `needs_root_rights` is not required on Pi 4B for correct KlipperScreen function.

alfrix, 2023-01-29:
the x-server sometimes needs roots rights sometimes it doesn't.
it usually depends on the video driver (if it's kms or not)
as this is a hw dependent it's not something that i should add by default (there are security concerns too)
so it's not a bug of klipperscreen but a something needed for some hw configurations

closed 2023-01-29 by alfrix