KlipperScreen/KlipperScreen #857: [BUG] Commit  9fa047e breaks Klipper serial interface on Bullseye

ReXT3D, 2022-12-21:
**Description**

Commit https://github.com/jordanruthe/KlipperScreen/commit/9fa047eb8beb3f503821b560e3364e3935f52bd2 triggers installation of 191 new packages on Pi OS Legacy (Buster).  One or more of these packages permanently affect Klipper's ability to correctly communicate with the MCU via the serial (USB) interface.  This is exhibited by Klipper "hanging" during `FIRMWARE_RESTART` or losing communication with the MCU shortly after a restart.

Klipper logs are filled with errors related to serial communication such as: `Got error -1 in read: (11)Resource temporarily unavailable`, `Timeout on wait for 'tmcuart_response' response` and similar.

For more information refer to my [Recent KlipperScreen commits break Klipper on Buster](https://klipper.discourse.group/t/recent-klipperscreen-commits-break-klipper-on-buster/5370) thread on Klipper Discourse.

**To Reproduce**

1. Install fresh image of `2022-09-22-raspios-buster-armhf-lite.img` from **raspberrypi.com**
2. Boot the Pi and install all Raspberry Pi OS updates
4. Install Klipper, Moonraker and Fluidd (or Mainsail) from their repective repos, either manually or with KIAUH
5. Verify that Klipper functions correctly and confirm that `FIRMWARE_RESTART` behaves normally
6. Install KlipperScreen manually via `git clone https://github.com/jordanruthe/KlipperScreen.git`
7. In `~/KlipperScreen/` switch to commit 82b8cbc via `git checkout 82b8cbcca5864c4b42ddd05a4e18000cc015c716`
8. Verify that Klipper functions correctly and confirm that `FIRMWARE_RESTART` behaves normally
9. In `~/KlipperScreen/` switch to commit 9fa047e via `git checkout 9fa047eb8beb3f503821b560e3364e3935f52bd2`
10. Confirm that Klipper no longer functions correctly by issuing `FIRMWARE_RESTART` that will either "hang" Klipper or cause loss of MCU communications
11. Unistall KlipperScreen
12. Confirm that Klipper still does not function correctly by issuing `FIRMWARE_RESTART` that will either "hang" Klipper or cause loss of MCU communications

This behaviour is particularly troubling since it appears like hardware malfunction and uninstalling KlipperScreen does not fix it.  A completely new SD card image is required with limited and controlled component reinstallation to avoid introducing the issue again.

Note that this bug does not appear when following the exact same steps on the more recent Bullseye release of Raspberry Pi OS.

ReXT3D, 2022-12-21:
One of the 191 "pre requisite" packages that are being installed and automatically started as part of this commit is "ModemManager".  It is also automatically started and it interferes with the serial interface function required by Klipper.  A workaround is to disable ModemManager as follows:

```
sudo systemctl disable ModemManager.service
```

Interestingly, I do not see ModemManager being installed on Bullseye.  The fundamental question remains why `libmpv-dev` pulls-in 191 pre-requisite packages on Buster?

Credits to Dragonkitty (replimat.eu) on Discord for describing the root cause of the issue.

closed 2023-01-29 by alfrix

thijstriemstra, 2024-08-19:
It looks like some KlipperScreen dependency installs ModemManager, I also found it on my system (and removed it).

commit eeffdd26 (2023-01-29): mask modemmanager close #857