KlipperScreen/KlipperScreen #858: Unusable GUI when switching between 2 printers that are both down (possible solution included)

Misterke, 2022-12-22:
**Describe the bug**
When having a config with 2 printers and both of them are disconnected at this time and neither is set as default, switching to the first works, but then going back to the selection screen and switching to the other fails (again on the list of power_devices)

**To Reproduce**
Have above mentioned configuration and just try switching between the printers.  See attached log on what happens.

**Expected behavior**
Correct switching between both printers.

**Logfile**
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10287311/KlipperScreen.log)

**Solution**
I think the cause is the current implementation of Printer.reset() in printer.py, which just puts every instance variable to None, even the stuff which is supposed to be a list / set.  Setting the majority of the instance variable to the same thing as in the __init__ method, (except for the state_cb, which I kept None), fixes this error at least.  Whether it addresses all problems will have to be seen in further testing.

Misterke, 2022-12-22:
I think I have something that works by removing the splash_screen panel when going to the printer_select panel.  The following patch seems to get me to something that does not hit any error and in the 2 minutes of testing so far seems to do what I would expect it to do ...
```diff --git a/screen.py b/screen.py
index 65f56a0..720083b 100755
--- a/screen.py
+++ b/screen.py
@@ -274,7 +274,11 @@ class KlipperScreen(Gtk.Window):

     def show_panel(self, panel_name, panel_type, title, remove=None, pop=True, **kwargs):
         try:
-            if remove == 2:
+            if remove == 3:
+                self._remove_all_panels()
+                if "splash_screen" in list(self.panels):
+                    del self.panels["splash_screen"]
+            elif remove == 2:
                 self._remove_all_panels()
             elif remove == 1:
                 self._remove_current_panel(pop)
@@ -615,7 +619,7 @@ class KlipperScreen(Gtk.Window):

     def show_printer_select(self, widget=None):
         self.base_panel.show_heaters(False)
-        self.show_panel("printer_select", "printer_select", _("Printer Select"), 2)
+        self.show_panel("printer_select", "printer_select", _("Printer Select"), 3)

     def process_busy_state(self, busy):
         self.process_update("notify_busy", busy)
```

Probably this is not the correct way to do this, but I guess someone more knowledgeable than me might get a hint on the correct thing to do from this ...

ressu, 2023-01-22:
This is caused by the `reset()` in ks_includes/printer.py. It sets every value to None.

https://github.com/jordanruthe/KlipperScreen/blob/b96adf55edd9389fc8f7e182a9104d790e86375a/ks_includes/printer.py#L10-L46

The way I fixed this is to move all of the initialisation from `__init__()` to `reset()` aside for `self.state_cb` and had init call reset instead. This makes the behaviour consistent regardless of whether we are starting from scratch or simply resetting the printer configuration.

PR #896 (2023-03-09): refactor printer switching
should fix #858 needs testing
prevent requests from  taking too long
closed 2023-03-09 by alfrix