KlipperScreen/KlipperScreen #859: Camera Panel doesn´t work as expected

gabber130779, 2022-12-23:
Today i tried to use camera panel on klipperscreen. I have uncommented the camera cofiguration in KlipperScreen. conf, but when I push the button on "Camera" on the screen, i get Message: no module named 'mpv'. what did i wrong?

robonxt, 2023-01-04:
I think you'll need to install the backend for the camera to work. Normally I would suggest using KIAUH to install the streamer thing, but it's been broken for a while. I just got mine working with a fresh installation of MainsailOS and enabling the camera panel

ckvsoft, 2023-01-28:
Even after configuring the camera, I can't see a camera button or anything similar.
what else is needed?

alfrix, 2023-01-29:
you need the version 0.3.1+  and have libmpv-dev installed on the system
monraker update manager could have failed to install the package and you didn't notice
if that's the case run: `sudo apt install libmpv-dev` and restart ks

the camera panel will not appear if the user have configured `use_default_menu = False`

closed 2023-01-29 by alfrix

ckvsoft, 2023-01-29:
Thx. It worket now. But i See only the Text, is there no icon?