KlipperScreen/KlipperScreen #862: [BUG] Memory Leak if ADB Android is locked

gamestailer94, 2022-12-27:
**Describe the bug**
I use an old Android as the Screen, as described [here](https://klipperscreen.readthedocs.io/en/latest/Android/) via USB(ABD).
When I lock the Android Phone, (because I go to bed) Klipper Screen starts using more and more memory, at about 1 MB / min.
It stops if I unlock the phone and some memory gets freed, but not all. (110MB before unlock, after unlock 91MB, normal is 68MB)
I did only test it while a print was running.

I created a memory usage log, using this [script](https://gist.github.com/gamestailer94/1e514825769f021a24ce469002aa5c86)

Android used, if relevant:
Samsung Galaxy Note Edge (SM-N915FY)
Android 8.1.0

**To Reproduce**
Connect Android as a Screen via USB ABD
Start monitoring the memory of klipper screen python script
Lock the Phone
Memory usage of script starts to increase

**Expected behavior**
To not use more memory

**Logfile**
[memory_usage.log](https://github.com/jordanruthe/KlipperScreen/files/10309436/memory_usage.log)
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10309443/KlipperScreen.log)

tomich, 2024-07-03:
Hi. One week ago I reinstalled everything and bumped into this bug. But I've been using this config for years without it. So I started checking my config. 

In the documentation there is a section with "Recommended configuration for this use case"

https://klipperscreen.readthedocs.io/en/latest/Android/#recommended-configuration-for-this-use-case

Are you sure you configured that?

In my case, whith use_dpms = False in KlipperScreen.conf and with the config I mentioned (but configured by hand via the KlipperScreen GUI) I stop having memory leaks. There is also another option for screen during prints and I also set it to Never.

What I think is happening is that KS's script is having to track the state of the screen and this has a ML in the code. But with this config, KS never tracks the status of the screen so it assumes it is always ON. 

Of course as a counter measure to screen damage you can set your phone/tablet config to turn off after X amount of time with no activity. 

Pros: No memory leak
Cons: You may need to manually touch physical buttons on your phone some times to turn the screen on. You can kinda fix this by adding a "adb shell input keyevent 26" command in some custom script, for instance, when you start printing. So the screen turns on when you start your prints. 

Let me know if this helped.

tomich, 2024-07-08:
An update on my las comment. 

With my changes you no longer get a "slow" memory leak that increments by the minute but you end up with something a lot worse. 

Sometimes when you wake the screen, other times randomly, memory on my host goes from ~15-25% to 99%. And it's all KlipperScreen. This has also happened when my printer is off and I turn it on. KS goes to 100% mem. 

I've had two failed prints already and this was the cause. 

Don't know why I was not having this issue before in my old installation(but updated). Maybe I was using an old version and locked to it without knowing. I remember changing a lot of things and having a custom launch script completely different from the example provided. I can't remember WHY I build a custom launch script but maybe it was related to this. 

This is kinda big because when I first encounter this bug it was about consumning a lot of memory but not all my mem. Now it makes prints fail. Kinda critical IMO. 

Can someone guide me a bit on how can I help with this bug? 

Now we have two situations:

1- When suspend screen/turn off screen is on: You get slow leak. And screen is accesible.
2- When screen is completely ON and you no longer track the state of the screen BUT phone screen turns off (and I assume phone goes to sleep) you dont get memory leak but if you wake phone you need to wait like 10 seconds for KS to be responsive there. Also. Randomly KlipperScreen will take ALL YOUR MEMORY.