KlipperScreen/KlipperScreen #869: [BUG] Virtual Keyboard for PSK Wifi is missing the ampersand "&" charactor

camerony, 2023-01-15:
**Describe the bug**
Go to join any WIFI Network and PSK entry keyboard is missing the & character.  So, if you use the & for your wifi password, it is not possible to join the network

**To Reproduce**
Steps to reproduce the behavior:
1. Touch Configuration
2. Touch Network
4. Touch the Arrow to join any new network
5. Touch to enter PSK for ssid
6. Inspect the keyboard layouts for the &

**Expected behavior**
That the ampersand "&" character is a possible keyboard character

**Screenshots**
[Screenshot](https://photos.app.goo.gl/GsrooUpaXJkK9Mai7)

**Logfile**
Attach your [klipperscreen.log](https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/)

JustTheDazz, 2023-01-19:
Having a similar issue where there is no % key doing the same thing

hawkeyexp, 2023-01-20:
A general hint - special chars like % and & are generally possible in a psk but in many many cases they are causing problems also in industrial environments - don't use them and you are more lucky :-)

camerony, 2023-01-21:
I've actually fixed this and will do a pull request after some more testing.  Fairly easy implementation.

closed 2023-01-21 by camerony