KlipperScreen/KlipperScreen #875: Invalid config file Option "use_dpms" not recognized for section "[printer FlsunQQ-S_PRO]"[BUG]

qkpony, 2023-01-23:
After I did a update on my klipper I have been getting this error just on my screen on the printer. Invalid config file Option "use_dpms" not recognized for section "[printer FlsunQQ-S_PRO]". It came after an update. I can't seem to figure it out. Now when I try to pen any of the config files through the web browser the circle just keeps spinning like it's waiting. but IDK why I am getting this on my screen. I hope I did this correctly.

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10477006/KlipperScreen.log)

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....

4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logfile**

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10477009/KlipperScreen.log)

qkpony, 2023-01-23:
I got it fixed. I just commented out the line for using dpms.

alfrix, 2023-01-29:
use_dpms goes in section [main]

it's one of the options that's preferable to use directly on the ui and not force it in the config

closed 2023-01-29 by alfrix