KlipperScreen/KlipperScreen #876: [BUG] RPI 3+ and BTT PI TFT V2.0 nor working with BULLSEYE debian. Stuck on terminal

vndk36, 2023-01-26:
**Describe the bug**
KlipperScreen was not working for me on my RPI 3+ A and B on a fresh install, while is was clearly working on my other printers.
After some digging, the only real diff was debian versions. Buster on the working printers and bullseye on the non working one.

I tried all the trouble shooting method from the troubleshooting guide of klipperscreen. Non of them worked.

Klipper screen was installed with KIAUH. I also tested it with the klipperscreen scripts only. 
I tested both ways of installing it on a fresh debian and on the Fluiddpi image. Same issue both times.

I think that the issue is related to some kind of access as I did not have any klipperscreen log either.

**To Reproduce**
Steps to reproduce the behavior:
1. Install a fresh install of bullseye lite (32bits) from raspberry pi imager or fluiddpi image 1.19.0
2. Install Klipper, moonraker, fluidd and then klipper screen through KIAUH. Or do the klipperscreen install with the install scripts
3. Now you should have the RPI that boots and have a working klipper install from the web UI. But the screen keeps on the terminal. 
4. Try the trouble shooting part of the klipperscreen wiki with no success

**Expected behavior**
Expected the screen to show the klipperscreen interface or at least to have some log as with buster

**Logfile**
No log files as it was not even generated.

**Fix**
Use raspberry imager to install Buster and install everything from KIAUH again

alfrix, 2023-01-29:
as said in the docs
> If KlipperScreen.log doesn't exist, run systemctl status KlipperScreen
(or journalctl -xe -u KlipperScreen)

https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/

JoHo94, 2023-02-12:
I have the same issue. Fresh MainsailOS with Debian 11.6 Bullseye. Here is the Log:
`Feb 12 18:59:02 mainsailos KlipperScreen-start.sh[2655]: Running KlipperScreen on X in display :0 by default
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: X.Org X Server 1.20.11
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: X Protocol Version 11, Revision 0
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: Build Operating System: linux Raspbian
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: Current Operating System: Linux mainsailos 5.15.76-v7+ #1597 SMP Fri Nov 4 12:13:17 GMT 2022 armv7l
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Composite-1:720x480@60i vc_mem.mem_base=>
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: Build Date: 20 December 2022  11:29:25PM
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: xorg-server 2:1.20.11-1+rpt1+deb11u4 (https://www.debian.org/support)
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: Current version of pixman: 0.40.0
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]:         Before reporting problems, check http://wiki.x.org
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]:         to make sure that you have the latest version.
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: Markers: (--) probed, (**) from config file, (==) default setting,
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]:         (++) from command line, (!!) notice, (II) informational,
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: (==) Log file: "/home/pi/.local/share/xorg/Xorg.0.log", Time: Sun Feb 12 18:59:03 2023
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: Parse error on line 1 of section InputClass in file /usr/share/X11/xorg.conf.d/99-fbturbo.conf
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]:         "/dev/fb0" is not a valid keyword in this section.
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: (EE)
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: Fatal server error:
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: (EE) no screens found(EE)
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: (EE)
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: Please consult the The X.Org Foundation support
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]:          at http://wiki.x.org
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]:  for help.
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: (EE) Please also check the log file at "/home/pi/.local/share/xorg/Xorg.0.log" for additional information.
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: (EE)
Feb 12 18:59:03 mainsailos KlipperScreen-start.sh[2659]: (EE) Server terminated with error (1). Closing log file.
`