KlipperScreen/KlipperScreen #881: [BUG]Showing error modal: Unable to load panel network Unable to create panel: network

findus4fun, 2023-01-29:
When trying to open the network panel I get:

Unable to create panel: network 

Logifle says: 
  File "/home/pi/.KlipperScreen-env/lib/python3.10/site-packages/NetworkManager.py", line 619, in to_python
    val['ipv4']['dns'] = [fixups.addr_to_python(addr,socket.AF_INET) for addr in val['ipv4']['dns']]
KeyError: 'dns'

**Logfile**
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10530440/KlipperScreen.log)

sabaatworld, 2023-01-30:
I'm seeing the similar issue while running Klipper and KlipperScreen on Odroid C4. Here's are the error logs:

```
2023-01-29 16:56:47,389 [screen.py:_load_panel()] - Loading panel: network
2023-01-29 16:56:47,390 [screen.py:_load_panel()] - Panel path: /home/sabaat/KlipperScreen/panels/network.py
2023-01-29 16:56:47,404 [network.py:__init__()] - Found wireless interfaces: ['wlan0']
2023-01-29 16:56:47,404 [network.py:__init__()] - Using NetworkManager
2023-01-29 16:56:47,587 [screen.py:_load_panel()] - 'dns'
Traceback (most recent call last):
  File "/home/sabaat/KlipperScreen/screen.py", line 270, in _load_panel
    return self.load_panel[panel](*args)
  File "/home/sabaat/KlipperScreen/panels/network.py", line 13, in create_panel
    return NetworkPanel(*args)
  File "/home/sabaat/KlipperScreen/panels/network.py", line 38, in __init__
    self.wifi = WifiManager(self.wireless_interfaces[0])
  File "/home/sabaat/KlipperScreen/ks_includes/wifi_nm.py", line 47, in __init__
    self._update_known_connections()
  File "/home/sabaat/KlipperScreen/ks_includes/wifi_nm.py", line 52, in _update_known_connections
    for con in NetworkManager.Settings.ListConnections():
  File "<string>", line 8, in ListConnections
  File "/home/sabaat/.KlipperScreen-env/lib/python3.10/site-packages/NetworkManager.py", line 578, in to_python
    val = fixups.base_to_python(val)
  File "/home/sabaat/.KlipperScreen-env/lib/python3.10/site-packages/NetworkManager.py", line 635, in base_to_python
    return [fixups.base_to_python(x) for x in val]
  File "/home/sabaat/.KlipperScreen-env/lib/python3.10/site-packages/NetworkManager.py", line 635, in <listcomp>
    return [fixups.base_to_python(x) for x in val]
  File "/home/sabaat/.KlipperScreen-env/lib/python3.10/site-packages/NetworkManager.py", line 648, in base_to_python
    return globals()[classname](val)
  File "/home/sabaat/.KlipperScreen-env/lib/python3.10/site-packages/NetworkManager.py", line 304, in __init__
    self.uuid = self.GetSettings()['connection']['uuid']
  File "<string>", line 8, in GetSettings
  File "/home/sabaat/.KlipperScreen-env/lib/python3.10/site-packages/NetworkManager.py", line 619, in to_python
    val['ipv4']['dns'] = [fixups.addr_to_python(addr,socket.AF_INET) for addr in val['ipv4']['dns']]
KeyError: 'dns'
2023-01-29 16:56:47,591 [screen.py:show_error_modal()] - Showing error modal: Unable to load panel network Unable to create panel: network
'dns'
2023-01-29 16:56:47,627 [KlippyGtk.py:Dialog()] - Showing dialog <Gtk.Dialog object at 0x7f690741c0 (GtkDialog at 0x55a7c2e830)>
```

[image]

mlwxxx, 2023-02-05:
I'm got the same errror too

alfrix, 2023-02-12:
see https://github.com/seveas/python-networkmanager/pull/94

PR #917 (2023-03-11): network: integrate networkmanager
fixes #881
this should be temporary until an implementation of sdbus-networkmanager
commit 094b90cf (2023-03-11): network: integrate networkmanager (#917)

fixes #881
this should be temporary until an implementation of sdbus-networkmanager
closed 2023-03-11 by alfrix