KlipperScreen/KlipperScreen #882: KlipperScreen Flashing between console and blank console

nimane9, 2023-01-30:
Hi,

I've been working through setting up KlipperScreen with my Raspberry Pi 3 Model B 1.2v from a Mainsail OS install. I had an issue where the screen would appear to default as a regular pi screen and just show the console log (I believe that's the term). It turned out to be an issue with virtual console permissions but I no longer get that error.  I had this same issue on an install of Fluidd on this same hardware but did a fresh install of Mainsail OS in an attempt to fix this.

It currently swaps between that screen and a blank console screen with just the blinking underscore every few seconds. I am following the instructions on the KlipperScreen troubleshooting page regarding the X11 configuration. When I run "cat /usr/share/X11/xorg.conf.d/99-fbturbo.conf | grep /dev/fb" I am met with an error stating there is "No such file or directory". I have ran "sudo apt-get update && sudo apt-get install xserver-xorg-video-fbturbo" but am met with the message "xserver-xorg-video-fbturbo is already the newest version (1.20210602~095927+1).".

KlipperScreen.log doesn't show up on Mainsail OS and KlipperScreen does not show up under the Update Manager, although the KlipperScreen folder does appear when accessing the files through an FTP connection.  I have attached the text provided upon running the "systemctl status KlipperScreen".

Thank you so much for the help! I know so little about linux but tried to do my due diligence before asking for help.

[KlipperScreen.txt](https://github.com/jordanruthe/KlipperScreen/files/10539956/KlipperScreen.txt)

miklschmidt, 2023-01-30:
Something seems to have screwed over a bunch of pretty important packages in a recent commit. Not sure exactly what happened, but this is really bad:
[image]

alfrix, 2023-02-05:
#883 not a bug of klipperscreen itself, sadly you wil have to reinstall

closed 2023-02-05 by alfrix