KlipperScreen/KlipperScreen #887: [BUG] Klipperscreen not getting correct display size

eboston, 2023-02-05:
**Describe the bug**
I am running a Manta M5P with the CB1 module.  I have a 7" HDMI Touchscreen with a resolution of 1024x600.  When I first plugged the display it, it appeared to be running at 1024x768.  I was able to change the resolution and the display is the correct size when booting.  But when KlipperScreen starts, the display is not the correct resolution.  The log shows it got a resolution of 1024x768.

**Expected behavior**
KlipperScreen to use the correct display size.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logfile**
--------------------KlipperScreen Log Start--------------------
Git Version: v0.2.8-36-g1950629
2023-02-03 14:15:36,806 [screen.py:main()] - KlipperScreen version: v0.2.8-36-g1950629
2023-02-03 14:15:36,809 [config.py:__init__()] - Config path location: /home/biqu/printer_data/config/KlipperScreen.conf
2023-02-03 14:15:36,833 [config.py:log_config()] -  ===== Config File =====

=======================
2023-02-03 14:15:36,835 [config.py:__init__()] - ====== Saved Def ======

=======================
2023-02-03 14:15:36,838 [config.py:__init__()] - Configuration validated succesfuly
2023-02-03 14:15:36,840 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
2023-02-03 14:15:36,941 [config.py:create_translations()] - Selected lang: None OS lang: en_US.UTF-8
2023-02-03 14:15:36,942 [config.py:install_language()] - Using system lang
2023-02-03 14:15:36,942 [config.py:install_language()] - Using lang en
2023-02-03 14:15:37,223 [screen.py:__init__()] - Screen resolution: 1024x768
2023-02-03 14:15:37,226 [KlippyGtk.py:__init__()] - img width: 69 height: 69
2023-02-03 14:15:37,786 [screen.py:_load_panel()] - Loading panel: splash_screen
2023-02-03 14:15:37,787 [screen.py:_load_panel()] - Panel path: /home/biqu/KlipperScreen/panels/splash_screen.py
2023-02-03 14:15:37,881 [screen.py:show_panel()] - Attaching panel splash_screen
2023-02-03 14:15:37,884 [screen.py:show_panel()] - Showing back. count: 0
2023-02-03 14:15:37,903 [screen.py:show_panel()] - Current panel hierarchy: ['splash_screen']

alfrix, 2023-02-05:
klipperscreen doesn't set resolution, that needs to be configured with the instructions of the manufacturer
on a raspberry pi it involves /boot/config.txt on some systems kernel commandline is needed
but again it's not klipperscreen

closed 2023-02-05 by alfrix

eboston, 2023-02-05:
I understand that Klipperscreen gets the resolution from the OS.  The problem is, the call to get the resolution is returning the wrong value.  As I said, the OS has the correct resolution of 1024x600, the the call in Klipperscreen is getting 1024x768, which is incorrect.