KlipperScreen/KlipperScreen #890: [BUG] GUI not showing, blue screen with "Launch these commands"

Stooovie, 2023-02-07:
KlipperScreen randomly doesn't work, showing a blue screen with following text instead of the GUI. The Xiaomi Redmi phone is not connected to the internet, so there was no update to it nor the SDL app. 

```
Launch these commands on your Linux PC:
export DISPLAY=192.168.0.210:0
export PULSE_SERVER=t cp: 192.168.0.210:4713 xfwm4 & firefox
To tunnel X over SSH, forward port 6000 in your SSH client
If you run Linux in chroot on this device, run: export DISPLAY=: 0 PULSE_SERVER=tc:127.0.0.1:4713
```
Running these commands on the Pi that runs KlipperScreen doesn't do anything. 

I was able to fix this in the past by uninstalling and reinstalling KS in KIAUH, but not this time. Both the Pi and the phone have been rebooted, didn't fix.

KS v0.3.1-17-g059f1a8

**Logfile**
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10672635/KlipperScreen.log)

alfrix, 2023-02-12:
> Running these commands on the Pi that runs KlipperScreen doesn't do anything.

those commands are not for klipperscreen and they will not work 
the guide is here: https://klipperscreen.readthedocs.io/en/latest/Android

problems with sdl setup/connection are not bugs of klipperscreen, ks is just an application

closed 2023-02-12 by alfrix

Stooovie, 2023-02-12:
I have followed the guide, that's how I had it working for half a year. Manually uninstalling and reinstalling KS did fix the issue for me this time. Thanks.

Stooovie, 2023-08-22:
Happening again for seemingly no reason.  Uninstall/reinstall of KS via KIAUH didn't fix it this time. Any tips?

Stooovie, 2024-06-07:
happening again after update via KIAUH. Uninstall/reinstall of KS didn't fix, back to the Blue screen with "launch these commands" in XDSL. Out of ideas. No hardware change, phone is rebooted, lsusb does show the phone in charge and debug mode