KlipperScreen/KlipperScreen #893: [BUG]

hawkeyexp, 2023-02-09:
**Describe the bug**
After Printer was auto powered off and waking up by power device with starting a print job from Mainsail KlipperScreen stays in home screen and doesn't swich to print screen.

Restarting the KlipperScreen service fixes the problem and it directly switches to print screen.

Log as attached starting with waking up the printer from KlipperScreen by powering on power device.

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10700349/KlipperScreen.log)

alfrix, 2023-03-01:
Uncaught exception <class 'gi.repository.GLib.GError'>: gdk-pixbuf-error-quark: 
Schwerer Fehler beim Lesen einer PNG-Bilddatei: bad adaptive filter value (0)

apparently this was caused by a corrupt thumbnail,
if you still have the gcode could you share it? and the png thumbnail generated too, it maybe bug of moonraker or just a corruption in the filesystem or something else.

i will code a safeguard against this case

gillesguillemin, 2023-03-03:
I've been experiencing what I believe is the same issue for quite a while actually. I've long thought that it was due to some regression with KlipperScreen but I'm realising the problem might be coming from some regression with some Cura update instead if it's related to corrupted thumbnails.

hawkeyexp, 2023-03-11:
Hi Alfrix, sorry for late reply (very high workload) - the thumbs are ok and after a klipperscreen restart shown without bugs, also the filesystem which i checked external additional to the default fsck which is performed on every boot (i configured it) - i had the effect in the past randomly on different jobs.
If i think about now an idea comes up which can possible interact: i have enabled skip object on moonraker side - possible an access problem to the job file while not finished by post processing ?

closed 2023-03-19 by alfrix

ArifBryan, 2023-05-07:
I have this issue too, if i start the print after powering on printer from mainsail, the job_status screen not displayed.
Logs attached.

Snippet from moonraker.log :
```
2023-05-07 20:34:50,462 [power.py:_schedule_firmware_restart()] - Power Device Printer: Klipper reports 'ready', aborting FIRMWARE_RESTART
2023-05-07 20:34:50,704 [job_state.py:_handle_started()] - Job state initialized: standby
2023-05-07 20:34:50,975 [klippy_connection.py:_check_ready()] - Klippy ready
2023-05-07 20:34:51,037 [common.py:build_error()] - JSON-RPC Request Error: 400
Connection already identified: {'name': 'OctoEverywhere', 'version': '2.5.5', 'type': 'other', 'url': 'https://octoeverywhere.com'}
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/common.py", line 541, in execute_method
    result = await callback(params)
  File "/home/pi/moonraker/moonraker/websockets.py", line 131, in _handle_identify
    raise self.server.error(
moonraker.utils.ServerError: Connection already identified: {'name': 'OctoEverywhere', 'version': '2.5.5', 'type': 'other', 'url': 'https://octoeverywhere.com'}
2023-05-07 20:34:51,465 [data_store.py:_init_sensors()] - Configuring available sensors: ['heater_bed', 'extruder']
2023-05-07 20:34:56,479 [job_state.py:_status_update()] - Job Started: Displaystuetze.gcode
```

Snippet from KlipperScreen.log :
```
2023-05-07 20:34:46,075 [screen.py:_websocket_callback()] - Power status changed: {'device': 'Printer', 'status': 'on', 'locked_while_printing': True, 'type': 'gpio'}
2023-05-07 20:34:50,980 [printer.py:change_state()] - Changing state from 'disconnected' to 'ready'
2023-05-07 20:34:50,980 [printer.py:change_state()] - Adding callback for state: ready
2023-05-07 20:34:50,985 [screen.py:close_screensaver()] - Closing Screensaver
2023-05-07 20:34:50,992 [screen.py:wake_screen()] - Screen wake up
2023-05-07 20:34:51,009 [main_menu.py:initialize()] - ### Making MainMenu
2023-05-07 20:34:51,114 [main_menu.py:add_device()] - Adding device: extruder
2023-05-07 20:34:51,125 [main_menu.py:add_device()] - Adding device: heater_bed
2023-05-07 20:34:51,137 [screen.py:attach_panel()] - Current panel hierarchy: main_panel
```

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/11415101/KlipperScreen.log)
[moonraker.log](https://github.com/jordanruthe/KlipperScreen/files/11415102/moonraker.log)

commit 34177652 (2023-03-19): job_status: fix panel not showing and not updating thumbnails

safeguard against thumbnail corruption or other errors fixes #893
better system to update name and thumbnails fixes #889