KlipperScreen/KlipperScreen #894: Touch not working with new mainsail os V1.0.0/V1.0.1 [BUG]

Treintjes, 2023-02-10:
**Describe the bug**
A clear and concise description of what the bug is.
Updated via clean install to 1.0 and 1.0.1 mainsailos and with both versions once klipperscreen is installed touch is inoperative tried modifying config, and via sudo neither lead to success of touch screen operation. moved back to mainsail 0.7.1 via clean install, installed klipperscreen and everything works fine... I will be posting this on mainsail-crew git as well as im not sure where the problem lies.

I suspect with moving from Buster to Bullseye there may have been a kernel/driver support for my screen dropped? I am using a BTT TFT70 v2.0 with dsi cable.

**To Reproduce**
Steps to reproduce the behavior:
perform clean install of newest mainsail ver. 1.0.1 and then follow with clean install of klipperscreen and attemp to utilize touch on klipperscreen. using BTT TFT70 V2.0
**Expected behavior**
normal operations of klipperscreen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logfile**
Attach your [klipperscreen.log](https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/)

Treintjes, 2023-02-11:
My mistake issue corrected

Treintjes, 2023-02-11:
Not klipperscreen or mainsail issue

closed 2023-02-11 by Treintjes

samuel235, 2023-02-18:
Hi @Treintjes - Are you able to confirm what this issue actually was just in case people like myself are making the same error/mistake but we come here thinking there is an issue with touch and the latest version?