KlipperScreen/KlipperScreen #901: (EE) no screens found(EE) on RPi3B+, Camera Module 4 and RPi 7" Screen

samuel235, 2023-02-16:
Hi,

I'm running RPi3B+ with 1GB of RAM and a Camera Module 3. I have a camera stream coming into MainsailOS via spyglass but this requires me to have `dtoverlay=vc4-kms-v3d` in my /boot/config.txt, this then causes KlipperScreen to not be displayed on my 7" official RPi touch screen. Upon checking my KlipperScreen journalctl I see the following:

```
Feb 16 19:39:34 raspberrypi systemd[1]: Started KlipperScreen.
░░ Subject: A start job for unit KlipperScreen.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit KlipperScreen.service has finished successfully.
░░
░░ The job identifier is 3937.
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1510]: Running KlipperScreen on X in display :0 by default
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: X.Org X Server 1.20.11
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: X Protocol Version 11, Revision 0
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: Build Operating System: linux Debian
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: Current Operating System: Linux raspberrypi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Com>
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: Build Date: 09 February 2023  03:15:54PM
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: xorg-server 2:1.20.11-1+rpt1+deb11u5 (https://www.debian.org/support)
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: Current version of pixman: 0.40.0
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]:         Before reporting problems, check http://wiki.x.org
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]:         to make sure that you have the latest version.
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: Markers: (--) probed, (**) from config file, (==) default setting,
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]:         (++) from command line, (!!) notice, (II) informational,
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: (==) Log file: "/var/log/Xorg.0.log", Time: Thu Feb 16 19:39:34 2023
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: (EE)
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: Fatal server error:
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: (EE) no screens found(EE)
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: (EE)
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: Please consult the The X.Org Foundation support
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]:          at http://wiki.x.org
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]:  for help.
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: (EE)
Feb 16 19:39:34 raspberrypi KlipperScreen-start.sh[1514]: (EE) Server terminated with error (1). Closing log file.
Feb 16 19:39:49 raspberrypi KlipperScreen-start.sh[1513]: /usr/bin/xinit: giving up
Feb 16 19:39:49 raspberrypi KlipperScreen-start.sh[1513]: /usr/bin/xinit: unable to connect to X server: Connection refused
Feb 16 19:39:49 raspberrypi KlipperScreen-start.sh[1513]: /usr/bin/xinit: server error
Feb 16 19:39:49 raspberrypi systemd[1]: KlipperScreen.service: Main process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ An ExecStart= process belonging to unit KlipperScreen.service has exited.
░░
░░ The process' exit code is 'exited' and its exit status is 1.
Feb 16 19:39:49 raspberrypi systemd[1]: KlipperScreen.service: Failed with result 'exit-code'.
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10760076/KlipperScreen.log)

░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ The unit KlipperScreen.service has entered the 'failed' state with result 'exit-code'.
```

I can confirm that the screen works properly and is registered with an ID of fb0. If I add the device ID into `/usr/share/X11/xorg.conf.d/99-fbturbo.conf` and restart KlipperScreen service it doesnt fix the issue, and then rebooting the system removes the file too.

I'm unable to comment out the `dtoverlay=vc4-kms-v3d` line of my /boot/config.txt file as this then breaks the camera stream.

Should you need any log files or anything please let me know. The KlipperScreen log file contains the same as the journalctl output mentioned above but I will attach a copy to this issue too.

alfrix, 2023-03-10:
there is not much you can do besides trying different configurations to see if helps, i really can't help since it's hardware specific, if you find a solution please add it here in case someone with the same issue find this.
Closing since KlipperScreen is just an application, the repo should track KS bugs and features, not hw specific things

NerdyGriffin, 2024-06-10:
@samuel235 I investigated this issue last year and have not yet found a solution. As I described in #1196:
> There is a known compatibility issue between KlipperScreen (or one of its dependencies) and the new camera stack "libcamera".
The Camera Module v3 is not compatible with the legacy camera stack, so (as of the time I am writing this) it requires libcamera to function.

For now, the only "solution" is to use an older model of the Camera Module that works with the legacy camera stack (such as the official v2 module or third-party brands) or to use a USB camera.
There are modules such as [this ArduCam](https://www.arducam.com/product/12mp-imx708-usb-uvc-102-wide-angle-fixed-focus-camera-module-3/) that would allow you to convert the camera module to USB, but I have not confirmed whether this is compatible with the Pi Camera Module 3 (Note the "Module 3" in that link is an ArduCam module, not the official Pi brand module)