KlipperScreen/KlipperScreen #905: [FR] Add Web UI access from display

Matszwe02, 2023-02-26:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Sometimes I need to go to my PC and access some settings that are not available on klipperscreen.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Add "WebUI" button to open a popup which has a browser with opened localhost with selected port (default 80). It ultimately opens Mainsail, Fluidd or OctoPrint

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Implementing a whole lot of functions in KlipperScreen (not as easy to implement)

**Additional context**
Add any other context or screenshots about the feature request here.

hawkeyexp, 2023-03-11:
Hm, does it make sense? Initial i think no - klipperscreen is an alternate control interface like mainsail fluid etc. - additional it is not really funny to control the web based interfaces on the small screens used on printers in most cases.

Matszwe02, 2023-03-12:
> Hm, does it make sense? Initial i think no - klipperscreen is an alternate control interface like mainsail fluid etc. - additional it is not really funny to control the web based interfaces on the small screens used on printers in most cases.

Right, but itself lacks many features. That webUI won't be for everyday use, just to control that one thing per month when you need for example to change pin in printer.cfg

alfrix, 2023-05-14:
duplicate of the idea in #682 the answer is: maybe

closed 2023-05-14 by alfrix