KlipperScreen/KlipperScreen #909: No macros in Klipperscreen

stenc55, 2023-03-01:
**Describe the bug**
No macros are shown in Klipperscreen. Armbian CLI on Orangepi 3LTS.
Only one showing is Load/Unload in Extrude screen.

**Expected behavior**
See macros when in macro screen

**Logfile**
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10866014/KlipperScreen.log)

alfrix, 2023-03-03:
can you please provide you printer.cfg or the file containing your macros sections?

stenc55, 2023-03-04:
Renamed to .txt, Github does not allow .cfg.
Also, in printer.config kiauh macros are included and can be started form Mainsail macro section
[image]

[kiauh_macros.txt](https://github.com/jordanruthe/KlipperScreen/files/10889621/kiauh_macros.txt)

alfrix, 2023-03-08:
i've tested [kiauh_macros.txt](https://github.com/jordanruthe/KlipperScreen/files/10889621/kiauh_macros.txt) and it works correctly, maybe the issue is inside `pa_tuner1.cfg` please attach it so i can test it.

stenc55, 2023-03-08:
[pa_tuner1.txt](https://github.com/jordanruthe/KlipperScreen/files/10923462/pa_tuner1.txt)

alfrix, 2023-03-10:
it is caused by the extra space at the end of the last two macros, any extra space at the start or end will cause this

commit dfe1ac8d (2023-03-11): macros: fix list not loading when there are extra spaces fixes #909
closed 2023-03-11 by alfrix