KlipperScreen/KlipperScreen #910: [BUG] latest update to bed_level.py removes "mid" screw position information in 3-screw bed

dctrjones, 2023-03-02:
**Describe the bug**
The latest update to `bed_level.py` fixed the screen scaling issues, Thank you!

However, now with a "3 screw" bed, the side with a single "mid" screw, is now reported in the interface as having a `None` position.  The icon is in the correct position but pushing the button will not move the extruder.  The side of the bed with 2 screws is found correctly and behaves as expected.

It appears that the new code at lines 110-113 will incorrectly find the "MID" screw and remove it from the list.  In my case the `fl=find_closest...`, will grab what is the mid-left screw, and remove it from the list.  

Later in the code when checking for `lm` and `rm` the `remaining_screws` variable is empty and the `find_closest` returns `None` for the `lm` screw.

In doing some (admittedly, rather crude) adjustments in `bed_level.py`, I was able to get it working again by bringing back the previous code for finding the first 4 corner settings.  Here's a snippet of what I put back:

```
107         remaining_screws = self.screws[:]
108         remaining_positions = []
109 
110 #        fl = find_closest(remaining_screws, (min_x, min_y), max_distance, remove=True)
111 #        bl = find_closest(remaining_screws, (min_x, max_y), max_distance, remove=True)
112 #        br = find_closest(remaining_screws, (max_x, max_y), max_distance, remove=True)
113 #        fr = find_closest(remaining_screws, (max_x, min_y), max_distance, remove=True)
114         
115         fl = [min_x, min_y]
116         bl = [min_x, max_y]
117         br = [max_x, max_y]
118         fr = [max_x, min_y]
119             
```
This ends up filling in a few extra screw positions, in this case an extra `fl` will be found and position information added.  But during the checks against the screw positions listed in `KlipperScreen.cfg` at lines 192-206, the extra `fl` screw is omitted and therefore the Bed Level screen displays and works as expected.

As mentioned, the above fix is a cludgy hack, but it works in my case!
 
**To Reproduce**
Steps to reproduce the behavior:

Set the bed_screw positions in `printer.cfg` as shown here:

`printer.cfg`
```
[bed_screws]
screw1: 7, 75
screw1_name: left-middle
screw2: 107, 25
screw2_name: front-right
screw3: 107, 115
screw3_name: back-right
```

`Klipperscreen.conf`
```
[main]
default_printer: Trinus

[printer Trinus]
screw_positions: lm, fr, br
```

The "left-middle" screw shown in the "Bed Level" screen will display as having a "None" value and will not move the bed when clicked.

**Expected behavior**
All screw icons will move the extruder position.

**Screenshots**
[image]

**Logfile**
My log file will show some "testing" outputs that will not appear with the standard code!

See log file at the 2023-03-01 21:21:06,919 time-stamp of the extra debugging added:

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10867353/KlipperScreen.log)

alfrix, 2023-03-03:
@czaky

closed 2023-03-08 by alfrix

czaky, 2023-03-09:
Sorry, for breaking this!
I may find some time this weekend to fix it.

[My plan was to add from 1 to 9 screw combinations anyway.]

How fast do you need it?

dctrjones, 2023-03-10:
For me (with 3 bed screws), this fix [926d78d](https://github.com/jordanruthe/KlipperScreen/commit/926d78de01efb114d00e021364b04bbe6f28690b) corrected the issue I had posted.

commit 926d78de (2023-03-08): bed_level: remove screws only if in list and simplify detection fixes #910