KlipperScreen/KlipperScreen #916: Save z offset as probe or endstop [FR]

Matszwe02, 2023-03-07:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
When I change z offset, klipperscreen asks me if I want to save it to probe, but I want to save it to endstop

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
2 buttons: "save to probe", "save to endstop"

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

alfrix, 2023-03-11:
this is already possible, it doesn't show the endstop option if the probe is the virtual endstop
i cannot check anything without your config or log

Matszwe02, 2023-03-11:
> this is already possible, it doesn't show the endstop option if the probe is the virtual endstop
> i cannot check anything without your printer config and klipperscreen log

Hi. Thanks for reply. I don't current have an access to my printer. It's delta, kinematics and klipperscreen config are default, with bed mesh probe. Is there an easy way to make the option to choose what to apply the offset to? I also don't always want to apply offset to endstops, but prove instead

alfrix, 2023-03-13:
> It's delta

that explains why, it forgot to add deltas in the mix.

danorder, 2023-03-22:
This wouldn't solve anything and, a whole community (flsun) uses a a  variation of Mentals zoffset macro to apply a slight babystep. but regardless delta calibrate would still be required to properly apply any "probe_calibrate" /zoffset changes. So other then confusing users Im not certain what would be solved since delta_calibrate still needs to run.

alfrix, 2023-03-22:
i don't have a delta, if someone can describe the proper procedure to apply the z_offset it would help me move this forwards faster

danorder, 2023-06-01:
Its probe_calibrate first. to get the "probe" offset if not already known. Then delta_calibrate to tram / apply the current offset.  If using a  probe   probes other then piezo or direct nozzle types  are barely recommended by kevin over using the manual paper method. due to "tilt" related issues.  To clarify my  previous response currently (babystep/gcode_offset) autosaves with the mental zoffset macro. but, many users are not aware of the recent mainsail button changes. These either save a new probe offset or,  apply a "endstop height override.  Currently the new endstop override just leads to either scraped beds later or printing "high" since users tend to forget they used a endstop override vs simply using the probe calibrate button with a delta calibrate to apply it.  The mental macro simply displays the current "gcode offset" applying it every reboot until changed.  if done right without the mental zoffset macro,  in theory the probe_calibrate value should be 0 or typically +/- 0.05ish.  Otherwise if values are higher then expected its typically related to the usual issues paper thickness , not waiting for thermal expansion , etc.  Once the probe offset is known this doesn't typically change delta_calibrate handles tramming from there. The only time this could change would involve  a nozzle change and, "external probe" otherwise the ideal piezo or nozzle direct type probes should just need "delta_calibrate to set any small height changes.  For clarity on "external" probes on delta this would be anything mounted on the effector plate" (kossel type)  These can work but, require "perfect" alignment and, wont' factor in any nozzle height changes since they just pop on with a magnet connected to the plate.

commit 7ba9e493 (2023-07-30): job_status: show save to endstop to apply offset for deltas close #916

Z_OFFSET_APPLY_ENDSTOP works in deltas since https://github.com/Klipper3d/klipper/commit/f078a279059670e692ea676237b152b4d42518e4
closed 2023-07-30 by alfrix