KlipperScreen/KlipperScreen #919: [BUG] Troubleshooting - GL Driver Section not available for ver. 11 (bullseye)

NikosSprocket, 2023-03-10:
Just an info. for updating the Wiki in the [Troubleshooting](https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/) section that Debian 11 (bullseye) does not have the GL Driver under the **_Advanced_** section of the **_raspi-config_**.

alfrix, 2023-03-10:
below the raspi-config mode there is the manual mode it says 

> Or:
> 
> manually edit

not a klipperscreen bug, you need to install the screen as instructed by the manufacturer
in general any screen is not working or touch is not working is not an issue with klipperscreen, remember klipperscreen is just an application, it doesn't control drivers.

closed 2023-03-10 by alfrix

NikosSprocket, 2023-03-10:
Sorry for the confusion. It's not a bug; just the documentation on the Wiki suggests that there is a "GL Driver" option under the Advance section of "raspi-config". This is not available since Debian bullseye. See below:

[image]

alfrix, 2023-03-11:
the docs need updating, but docs are correct the menu was still there in the pi3

NikosSprocket, 2023-03-11:
Wow! I have Pi4s only and it's not!!! I will never have guest that 3 and 4 has that many differences. Good to know.

```
:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
:~ $ uname -a
Linux voron24-01 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
```

[image]

Below is the main menu from my 4 (