KlipperScreen/KlipperScreen #921: [FR] ultra wide screen support.

LazaroFilm, 2023-03-10:
**Is your feature request related to a problem? Please describe.**
I'm considering using an ultrawide touchscreen with my printer but it doesn't seem like KS has an option to support ultra wide screens. 

**Describe the solution you'd like**
Either implement a preset template for ultrawide screens or add a layout customization option. 

**Describe alternatives you've considered**
Using Mainsail instead but that's not touch oriented either but allows for UI layout customization 

**Additional context**
Link to ultra wide screen I'm referring to (there are many like this) 
https://a.aliexpress.com/_mOjcX3Y

ulope, 2023-10-17:
I've also just discovered this display. 
There's [a post without much detail in the Klipper discourse](https://klipper.discourse.group/t/hardware-known-to-work-with-klipperscreen/35/57) that seems to show the display working.

ulope, 2023-10-24:
I just went ahead and bought it - turns out it works perfectly fine with KS already!
[image]

Looks like the docs need updating.

Nathan22211, 2024-10-05:
TBH it looks like you could shrink the temp display on something this size.