KlipperScreen/KlipperScreen #923: Extruder Temperature not adjustable[BUG]

xArgentiusx, 2023-03-11:
After updating to Klipperscreen V0.3.2-11 I can no longer set the extruder temperature on my display. The printer doesn't respond to it. However, the heating bed and chamber can be changed. The preheat settings don't work either and the cooldown also not working. All other menu items work. only the extruder temperature does not work in any menu. in the browser interface everything works without problems.

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10948765/KlipperScreen.log)

hapklaar, 2023-03-11:
Same issue here. I get the following entry in KlipperScreen.log

```
2023-03-11 21:12:20,849 [functions.py:logging_exception_handler()] - Uncaught exception <class 'KeyError'>: 'max_temp'
  File "/home/pi/KlipperScreen/panels/temperature.py", line 233, in set_temperature
    max_temp = float(self._printer.get_config_section(heater)['max_temp'])
NoneType: None
```

hapklaar, 2023-03-11:
The temporary fix suggested in another thread for other issues with this build, also works here:

https://github.com/jordanruthe/KlipperScreen/issues/924#issuecomment-1464999967

xArgentiusx, 2023-03-11:
https://github.com/jordanruthe/KlipperScreen/issues/924#issuecomment-1465023187

this one helps with my issue

commit 87c3f037 (2023-03-11): fix issues caused by dfe1ac8d

fix #924
fix #923
closed 2023-03-11 by alfrix