KlipperScreen/KlipperScreen #924: Unable to create panel: job_status[BUG]

ivvi25, 2023-03-11:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logfile**
Attach your [klipperscreen.log](https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/)

ivvi25, 2023-03-11:
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10949064/KlipperScreen.log)
[image]

DeGrandNagus, 2023-03-11:
Same issue here, OP might wanna fill out the form though.

Anyway, this happens when you start a print. Everythings fine before then.

Geeps-SA, 2023-03-11:
Exact same thing my side but also noticed it on Z_calibrate 
[image]
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10949168/KlipperScreen.log)

ivvi25, 2023-03-11:
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/10949182/KlipperScreen.log)

robertspark, 2023-03-11:
I have a very similar message

Unable to create panel: move 'kinematics' 

when going to the move screen, i had just updated klipperscreen today :(

I presume they are all related.

ivvi25, 2023-03-11:
Yes, i had just updated klipperscreen today.

ValterFerreira, 2023-03-11:
> I just installed -3 commits of the github repo and it seems to work fine (the issue arises at -2 commits, (dfe1ac8) v0.3.2-10
> 
> {if you want a fix go into terminal via ssh and type:
> cd KlipperScreen
> git checkout be473bf
> ./scripts/KlipperScreen-install.sh
> 
> and it should work fine as a temp fix

I tried but it doesn't work for me, the error remains.

robertspark, 2023-03-11:
try 

cd KlipperScreen
git checkout HEAD~5
./scripts/KlipperScreen-install.sh

note the ~ is the tide symbol

is it ACTUALLY installing or failing at some point?

commit 87c3f037 (2023-03-11): fix issues caused by dfe1ac8d

fix #924
fix #923
closed 2023-03-11 by alfrix