KlipperScreen/KlipperScreen #928: [BUG] My touch did not work after Update to V0.3.2-13-gd6fbdc

Andi3D-tech, 2023-03-13:
I use the BTT PiTFT50 without problems.
After the update, i can see klipperscreen on the display, but touch did not work anymore.
I use MainsailOS 32bit Bullseye.
Klipper v0.11.0-128
Moonraker v0.8.0-29
Mainsail v2.5.0

AP4130, 2023-03-13:
Same issue here, with the same hardware (on an RPi 4). I have verified the touchscreen touch function works without issue on a different Pi running another OS. 

There appears to be a building Reddit thread on the issue, too:
https://www.reddit.com/r/klippers/comments/11m0678/klipperscreen_touch_stopped_working/?utm_source=share&utm_medium=ios_app&utm_name=iossmf

Andi3D-tech, 2023-03-14:
Follow this and it will work again!!!

https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/#touch-not-working-on-debian-bullseye

AP4130, 2023-03-14:
Confirmed on mine as well! I'm an idiot- I saw those steps on the Klipperscreen site, but didn't think to try because my screen had been working without issue on Bullseye previously.

Andi3D-tech, 2023-03-14:
> Confirmed on mine as well! I'm an idiot- I saw those steps on the Klipperscreen site, but didn't think to try because my screen had been working without issue on Bullseye previously.

You are not an idiot!!!
It works on mine as well without problems.
Dont know what happens during the update...

hmouridsen, 2023-03-14:
I've just reinstalled Mainsail and Klipperscreen - witch means upgrade to Bullseye as well.

Does this issue also freeze up the display? Not just touch is unresponsive but screen freeze up also 😕

My screen is the BTT tft50

virtualcomputerguy, 2023-03-15:
Well that was fun. I did not have the option to select G2 from the pi config. Guess I will look at that later.
Had to manually edit the - /boot/config.txt

From ssh - sudo nano /boot/config.txt
You can then find the line - dtoverlay=vc4-kms-v3d
change it to dtoverlay=vc4-fkms-v3d

Save the file and reboot everything. Touch screen is working. 

Seems like this is from a bad update...?? Mine was working until a small update yesterday then it all went bad. Had to reload Klipper and then get the touch screen working again.

dctrjones, 2023-03-18:
I had a similar issue after performing all updates on my RPI (Debian Bullseye) through Mainsail (including OS update).  The screen is a Waveshare 4.3", Klipper screen displays correctly but now all touch inputs only work on the left side of the screen.  

It's as if the touch sensitive area has been scaled by 0.5 on X.  For example to click the far right button on the screen I have to click just left of the centre of the screen.

I went through the troubleshooting steps using `raspi-config`, but didn't find the "FKMS" option there so had to edit the `/boot/config.txt` file to add the appropriate `dtoverlay=vc4-fkms-v3d`

To correct the "half screen" touch input I followed the instructions for ROTATING the screen [here](https://klipperscreen.readthedocs.io/en/latest/Hardware/).  To find the touch input device I ran:

`DISPLAY=:0 xinput`

With the following results:
`⎡ Virtual core pointer                    	id=2	[master pointer  (3)]`
`⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]`
`⎜   ↳ raspberrypi-ts                          	id=7	[slave  pointer  (2)]`
`⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]`
`   ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]`
`    ↳ UVC Camera (046d:0825)                  	id=6	[slave  keyboard (3)]`

But instead of INVERTING the touch area I had to scale it using using the following command:

`DISPLAY=:0 xinput set-prop 'raspberrypi-ts' 'Coordinate Transformation Matrix' 0.5 0 0 0 1 0 0 0 1`

NOTE: in the above command 'raspberry-ts' may NOT be the same on your system.

To make that change persistent after a reboot, I followed the instructions listed at the bottom of the article, editing the `/usr/share/X11/xorg.conf.d/40-libinput.conf` file.  This seems to have fixed my issue, for the most part, not perfectly accurate when using the smaller keyboard on the screen, but passable for the bigger buttons.

Get0ffMyLawnV2, 2023-03-18:
>Thank you very much for this.  Been at this for far too long and this worked for me.  I really appreciate it!

dctrjones, 2023-03-19:
> Thank you very much for this. Been at this for far too long and this worked for me. I really appreciate it! Edit: To be more specific, I'm running a BTT TFT50 and the buttons seem to line up just fine now with touch.

Glad it helped!  I was surprised by the change in the touchscreen after updating Bullseye.

Eclypse3D, 2023-03-19:
Yeah my bud has the same issue with his BTT TFT50 V2, he got the touch working but everything is off matrix wise.  I have 3 other units myself two Pi based BTT TFT50 V2 and a TFT43 on my trident and my SPI TFT35 on my other Trident.  The SPI screen works fine but now it does not wake up from sleep till the print ends then the touch screen no longer works so that is a slightly different behavior to the DSI screens.  I have not updated the other 2.4 or Trident as they have been offline for a few days while I finished building the Trident with eh SPI screen, so now I am not going to touch those two update wise I think till there is a functioning patch fix for it.  :)

harlock3, 2023-03-24:
New install PD5 with CM4 (emmc version). Touch screen doesn't work.  Printer isn't connected yet but I assume screen should respond to touch when I press on kipper restart, firmware restart or Menu icons. Tried the fix mentioned but no change. Using mainsail; all software installs (pi, klipper, mainsail, etc) on current versions

Eclypse3D, 2023-03-24:
Simply comment out the kms overlay line altogether if changing it does not work.

harlock3, 2023-03-24:
> Simply comment out the kms overlay line altogether if changing it does not work.

no change

Nephiel, 2023-04-02:
For the record, I got [this 5-inch DSI touchscreen](https://github.com/Freenove/Freenove_Touchscreen_Monitor_for_Raspberry_Pi) to work thanks to the information in this issue.

1. I _did_ need to set `dtoverlay=vc4-fkms-v3d` in `/boot/config.txt` by hand, since the `raspi-config` menu did not show the right option. Without this, the touch input does not work, and the error `rpi_touchscreen_attiny 10-0045: Failed to read REG_ID reg: -5` pops up in `dmesg`. After a reboot, KlipperScreen did not start,
2. I _did_ need to `sudo bash -c "echo needs_root_rights=yes>>/etc/X11/Xwrapper.config"` as per the [troubleshooting docs](https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/#cannot-open-virtual-console), otherwise KlipperScreen didn't start and the error `xf86OpenConsole: Cannot open virtual console 2 (Permission denied)` popped up in `/var/log/Xorg.0.log`.
3. I _did_ need to set the calibration matrix using `echo 'ACTION=="add", ATTRS{name}=="raspberrypi-ts", ENV{LIBINPUT_CALIBRATION_MATRIX}="0.5 0 0 0 1 0 0 0 1"' | sudo tee /etc/udev/rules.d/51-touchscreen.rules` as per [dctrjones' comment](https://github.com/jordanruthe/KlipperScreen/issues/928#issuecomment-1475026572) and [the hardware docs](https://klipperscreen.readthedocs.io/en/latest/Hardware/#touchscreen-touch-rotation), otherwise the touch input area was scaled to 50% horizontally and only responded to touches on the left half of the screen.
- I did _not_ need to enable I2C nor SPI for either the screen or the touch input to work.

KennyBeaudoin, 2023-04-12:
> Follow this and it will work again!!!
> 
> https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/#touch-not-working-on-debian-bullseye

it does not even show the FKMS anymore in the new updates.... 
and if you manually add it goes to a blank screen

KennyBeaudoin, 2023-04-13:
> pi@DELTABUSTER:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.1.21+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (                    Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #164                    2 Mon Apr  3 17:19:14 BST 2023
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr                    =00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruc                    tion cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] random: crng init done
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Reserved memory: created CMA memory pool at 0x0bc00000, size 256                     MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id share                    d-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835                    .enable_headphones=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:480x320M@59D smsc9                    5xx.macaddr=B8:27:EB:6E:41:82 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000                    000  console=ttyS0,115200 console=tty1 root=PARTUUID=81272009-02 rootfstype=ext4                     fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, l                    inear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, li                    near)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 177016K/458752K available (8985K kernel code, 1450K rwdat                    a, 2772K rodata, 432K init, 589K bss, 19592K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 33370 entries in 98 pages
[    0.000000] ftrace: allocated 98 pages with 3 groups
[    0.000000] trace event string verifier disabled
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000005] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2                    147483647500ns
[    0.000068] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_                    idle_ns: 1911260446275 ns
[    0.000152] bcm2835: system timer (irq = 27)
[    0.000970] Console: colour dummy device 80x30
[    0.001518] printk: console [tty1] enabled
[    0.001602] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060288] pid_max: default: 32768 minimum: 301
[    0.060503] LSM: Security Framework initializing
[    0.060801] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linea                    r)
[    0.060875] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes,                     linear)
[    0.062373] cgroup: Disabling memory control group subsystem
[    0.062776] CPU: Testing write buffer coherency: ok
[    0.064317] cblist_init_generic: Setting adjustable number of callback queues                    .
[    0.064391] cblist_init_generic: Setting shift to 0 and lim to 1.
[    0.064624] cblist_init_generic: Setting shift to 0 and lim to 1.
[    0.064929] Setting up static identity map for 0x8220 - 0x8258
[    0.066189] devtmpfs: initialized
[    0.078758] VFP support v0.3: implementor 41 architecture 1 part 20 variant b                     rev 5
[    0.079148] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, ma                    x_idle_ns: 19112604462750000 ns
[    0.079236] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.115642] pinctrl core: initialized pinctrl subsystem
[    0.117563] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.120661] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.126987] audit: initializing netlink subsys (disabled)
[    0.128469] thermal_sys: Registered thermal governor 'step_wise'
[    0.129239] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.129327] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.129569] Serial: AMBA PL011 UART driver
[    0.130627] audit: type=2000 audit(0.120:1): state=initialized audit_enabled=                    0 res=1
[    0.137501] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.160709] raspberrypi-firmware soc:firmware: Attached to firmware from 2023                    -03-17T10:52:42, variant start
[    0.170734] raspberrypi-firmware soc:firmware: Firmware hash is 82f3750a65fad                    ae9a38077e3c2e217ad158c8d54
[    0.221238] kprobes: kprobe jump-optimization is enabled. All kprobes are opt                    imized if possible.
[    2.409473] bcm2835-dma 20007000.dma: DMA legacy API manager, dmachans=0x1
[    2.412062] SCSI subsystem initialized
[    2.412551] usbcore: registered new interface driver usbfs
[    2.412692] usbcore: registered new interface driver hub
[... cut ...]
[   36.586768] Bluetooth: BNEP socket layer initialized
[   36.640028] Bluetooth: MGMT ver 1.22
[   36.785642] NET: Registered PF_ALG protocol family
[   37.065414] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   39.518142] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  150.320118] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net                    .ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instea                    d
pi@DELTABUSTER:~ $ reboot

alfrix, 2023-10-20:
closing since this fixes have been added to the docs

if more issues appear under bookworkm (not released for raspberry yet) then it should be on another thread

closed 2023-10-20 by alfrix