KlipperScreen/KlipperScreen #937: [FR] Possibilité de quitter ou réduire KlipperScreen

Michaeldu28, 2023-03-27:
Bonjour,
Merci beaucoup pour cette application.
Je souhaiterai savoir s'il serait possible d'avoir l'application en fenêtrée afin d’accéder au bureau de l'OS.
Un bouton quitter sur la page de démarrage serait également intéressant afin d'éviter de forcer l'arrêt du service.
Actuellement sur armbian, l'application démarre dès le démarrage du système et je ne peux donc choisir d'aller sur le bureau du système notamment pour accéder à Octoprint.

Merci d'avance pour votre réponse et votre éventuelle amélioration.
Cordialement

alfrix, 2023-04-01:
please write in english, i dont speak french, from the translator i can tell that you want klipperscreen to start windowed, 
this is possible already, but needs a more specific setting for people to use
you can disable ks service if you don't want it to start on boot.
klipperscreen is not compatible with Octoprint.
https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/#octoprint

alfrix, 2023-04-12:
closing since start windowed is part of #822 and the rest was answered

closed 2023-04-12 by alfrix