KlipperScreen/KlipperScreen #942: [BUG] 3.5" touchscreen went blank after an update

INeDiA, 2023-04-01:
Everything was fine since a week ago, but after I did a full update (both KlipperScreen and an _apt upgrade_) via kiauh, and then a reboot, my screen went dark; I can only see a dash "_" in the upper left corner.
I tried almost everything, from reinstalling drivers (LCD-show) to reinstalling KlipperScreen to fidgeting with configuration edits found everywhere: same results, screen is always black.
I'd rather not do a fresh install, I have a quite complex setup with multiple octoprint and klipper instances and it would take ages.

The screen is not broken, I can see things while the rpi boots.

**Logfile**
[KlipperScreen.log.zip](https://github.com/jordanruthe/KlipperScreen/files/11129721/KlipperScreen.log.zip)
The weird thing is that the last update was on March 24th, the day everything stopped working.

[image]

Thanks in advance for your help!

qkpony, 2023-04-01:
I had this same issue when I updated 2 weeks ago. Everything updated good. klipperscreen was on the klipper reboot menu I clicked the screen it rebooted I seen the rainbow screen and now it just black with the backlight still on. I have tried everything. I will be watching this.

INeDiA, 2023-04-05:
I'm not sure this is related to KlipperScreen anymore; I tried to show the regular desktop and the same happens, so maybe some other kind of update broke the config and/or the driver

napalmz, 2023-04-11:
I was able to fix this removing a package:
`sudo apt purge xserver-xorg-video-fbturbo`

Then restarted and it was all ok
`sudo systemctl restart KlipperScreen.service`

And after a reboot is still working.

Source: https://bytemeta.vip/repo/jordanruthe/KlipperScreen/issues/349

INeDiA, 2023-04-11:
> I was able to fix this removing a package: `sudo apt purge xserver-xorg-video-fbturbo`
> 
> Then restarted and it was all ok `sudo systemctl restart KlipperScreen.service`
> 
> And after a reboot is still working.
> 
> Source: https://bytemeta.vip/repo/jordanruthe/KlipperScreen/issues/349

oh man I love you so much
can confirm, it is now working