KlipperScreen/KlipperScreen #944: [BUG]

colly05, 2023-04-03:
Not sure if this is a bug or cfg issue. But the Home Screen I’m looking to get the colour showing on all temperatures 
[image]
All others work fine and even octopus works to hide the colour on the graph but it doesn’t show colour as the pic.

colly05, 2023-04-15:
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/11240307/KlipperScreen.log)

alfrix, 2023-04-16:
i don't see any errors does other themes work?

colly05, 2023-04-16:
there is no issues apart from the colour not showing if i remove say the  ebb36 in printer.cfg it comes back. so im unsure if its an issue with how many items ? the colour shows on the graph fine and goes like the others but does not show next to it ?
thanks

colly05, 2023-05-09:
it seams to follow which is last in the printer.cfg ?

commit 5a0df3a6 (2023-06-22): fix colors not applying correctly fix #944
closed 2023-06-22 by alfrix