KlipperScreen/KlipperScreen #948: Klipper SCREEN not Working ... touch Also on others [BUG]

KennyBeaudoin, 2023-04-10:
**Describe the bug**
i cannot see the FKMS option anymore to enable touch .. ive tried multiple stuff such as reinstall the os .. change os to other such as lite and the touch does not work at all ... ive tried manually adding the FKMS in the boot config file and it puts the screen white or black depending on the changes i make in FB0 or FB1 .. ive been on this issue for 4 days in a row and still cant figure out why the screens dont work .. i have a tft50 from btt and the other one is a 3.5 waveshare clone .. one is on a rpi zero and the other one on a 3b 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logfile**
Attach your [klipperscreen.log](https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/)

KennyBeaudoin, 2023-04-11:
pi@5PLUSs:~ $ systemctl status KlipperScreen
● KlipperScreen.service - KlipperScreen
     Loaded: loaded (/etc/systemd/system/KlipperScreen.service; enabled; vendor>
     Active: active (running) since Mon 2023-04-10 20:56:40 EDT; 1s ago
   Main PID: 4606 (KlipperScreen-s)
      Tasks: 3 (limit: 2057)
        CPU: 76ms
     CGroup: /system.slice/KlipperScreen.service
             ├─4606 /bin/bash /home/pi/KlipperScreen/scripts/KlipperScreen-star>
             └─4609 /usr/bin/xinit /home/pi/.KlipperScreen-env/bin/python /home>

Apr 10 20:56:40 5PLUSs KlipperScreen-start.sh[4610]: (EE)
Apr 10 20:56:40 5PLUSs KlipperScreen-start.sh[4610]: Fatal server error:
Apr 10 20:56:40 5PLUSs KlipperScreen-start.sh[4610]: (EE) no screens found(EE)
Apr 10 20:56:40 5PLUSs KlipperScreen-start.sh[4610]: (EE)
Apr 10 20:56:40 5PLUSs KlipperScreen-start.sh[4610]: Please consult the The X.O>
Apr 10 20:56:40 5PLUSs KlipperScreen-start.sh[4610]:          at http://wiki.x.>
Apr 10 20:56:40 5PLUSs KlipperScreen-start.sh[4610]:  for help.
Apr 10 20:56:40 5PLUSs KlipperScreen-start.sh[4610]: (EE) Please also check the>
Apr 10 20:56:40 5PLUSs KlipperScreen-start.sh[4610]: (EE)
Apr 10 20:56:40 5PLUSs KlipperScreen-start.sh[4610]: (EE) Server terminated wit>

KennyBeaudoin, 2023-04-12:
Version: v0.11.0-169-g83308a10
OS: Raspbian GNU/Linux 11 (bullseye)
Distro: MainsailOS 1.1.1 (bullseye)

alfrix, 2023-04-12:
this kind of issues are not not related to KS see #928

closed 2023-04-12 by alfrix

KennyBeaudoin, 2023-04-12:
> this kind of issues are not not related to KS see #928

please like i replied in other post ... this link info is not valid or good ... it does not work anymore ive tested it on several printers i believe that some missing coding or some updates is making it not work