KlipperScreen/KlipperScreen #949: [BUG] Touch input not working with BTT PI TFT50

BorisBrock, 2023-04-12:
**Describe the bug**
I have installed Mainsail OS and KlipperScreen on a Raspberry Pi 3b. Also I have attached a BTT PI TFT50 touch display, which should be supported out of the box by KlipperScreen. All software (OS, Mainsail, KS etc.) are up-to-date.

The display works fine, except touch input is not working at all. No matter where I touch the screen, nothing happens.

**To Reproduce**
Steps to reproduce the behavior:
Set up a system as described above.
Touch is not working.

**Expected behavior**
I would expect touch input to work properly.

**Logfile**
Attach your [klipperscreen.log](h[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/11208539/KlipperScreen.log)/)

closed 2023-04-12 by alfrix

KennyBeaudoin, 2023-04-12:
> not related to KS see #928

the link you are providing is not working anymore .. like what they say to do in that page does not work at all ive tested it on 2 different printers and spent over 60hours on them trying to make it work .. i am willing to jump to discord and do it with you guys and show you to find the problem .. but ive posted this on mainsail too and they said this is related to Klipper screen and sent me here ... now you are telling me this is not related to KS ... so its related to what ?? where can i actually get help and find the issue ? thanks a lot

BorisBrock, 2023-04-13:
> the link you are providing is not working anymore .. like what they say to do in that page does not work at all ive tested it on 2 different printers and spent over 60hours on them trying to make it [...]

Yesterday I was able to fix this issue on my machine (Raspberry Pi 3B). This is what I did:

1. Call `sudo nano /boot/config.txt`
2. Change `dtoverlay=vc4-kms-v3d` to `dtoverlay=vc4-fkms-v3d`
3. Change `max_framebuffers=2` to `max_framebuffers=1`
4. Exit and save
5. Call `sudo bash -c "echo needs_root_rights=yes>>/etc/X11/Xwrapper.config"`
6. Call `reboot`

After that the screen is working reliably and touch input is fully functional (at least for me).

KennyBeaudoin, 2023-04-13:
i have a 3.5 screen and it did not work this morning ... ill try on my tft 50 with my 3b tonight

Grut3d, 2023-04-17:
Thanks Boris, that worked for me!

gundalfatm, 2023-05-02:
Thanks @BorisBrock! Had the same issue with Pi 3B+ and Waveshare 4.3. Seems to work now :-D

XIF0rce, 2024-10-20:
YEEEEES, thanks @BorisBrock . Worked!!!!!!!!!!!!